armbian/build #6518: I would like to ...

jackeyt, 2024-04-23:
### Which feature would you like to have?

I had saw a [PR](https://github.com/armbian/build/pull/6256/files) for building a frameware package  for mobile phone: Redmi K30 Pro,which has 
 CPU [SG865],or called QCS8250.
I used to flash img file to devices throughing a tool,called QFIL ,which runnig in Windows PC. it downloads the image files by using USB,port:9008。
SO,in Armbian Project,how can I build a image file and flashing it to SG865 Machine,such as :
Redmin K30Pro or XiaoMi10.Is that any readme note?

### Funding

- [X] [I am willing to cover expenses](https://www.armbian.com/contact/)

amazingfate, 2024-04-24:
Armbian is providing sm8250(snapdragon  865/870) images bootable with qcom's abl loader, which contains one abl kernel image and one rootfs image. You can see flashing guide for xiaomi pad5 pro: https://github.com/amazingfate/armbian-xiaomi-elish/wiki/Flashing-Guide.

prahal, 2024-05-14:
@SteeManMI 
> Closing.

this bug about asking how to build and flash is not closed. Did you intend to close it or ask for one to close it?

SteeManMI, 2024-05-14:
> @SteeManMI
> 
> > Closing.
> 
> this bug about asking how to build and flash is not closed. Did you intend to close it or ask for one to close it?

I intended to close it.  This is not bug/issue with the build framework therefore this is not the appropriate place for this.  Use the forums.

closed 2024-05-14 by EvilOlaf