armbian/build #6527: Ubuntu 24.04 as host is unsupported

xlazom00, 2024-04-26:
### What happened?

Installing host-side packages [ python3-distutils ]
E: Package 'python3-distutils' has no installation candidate

https://packages.ubuntu.com/mantic/python3-distutils

python3-distutils ended in 23.10 :(

### How to reproduce?

run ./compile on ubuntu 24.04

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-04-26:
Noble is not yet officially supported ... but yeah, problems of this kind are expected.

xlazom00, 2024-04-26:
https://bugs.launchpad.net/ubuntu/+source/python3-stdlib-extensions/+bug/2060772

rpardini, 2024-04-28:
Feel free to test https://github.com/armbian/build/pull/6532

PR #6532 (2024-04-28): hostdep fixes for Noble distutils and pyelftools removed hooks
#### hostdep fixes for Noble distutils and pyelftools removed hooks

- prepare-host.sh: deps: replace `liblz4-tool` with `lz4`
  - should fix sid and trixie, which dropped the metapackage
- prepare-host.sh: deps: Noble dropped `python3-distutils`; fixes #6527
  - keep it for older releases
- prepare-host.sh: deps: add `libgnutls28-dev`, needed for some u-boot builds
- multiple boards: drop hooks adding 'python3-pyelftools' & `libgnutls28-dev` hostdeps, now included in standard core deps
commit 6a242ab1 (2024-04-28): prepare-host.sh: deps: Noble dropped `python3-distutils`; fixes #6527

- keep it for older releases
closed 2024-04-28 by igorpecovnik