armbian/build #6561: Builds fail for EOS distributions (for example focal)

SteeManMI, 2024-05-08:
### What happened?

This was reported in the forums: [https://forum.armbian.com/topic/34923-csc-armbian-for-rk322x-tv-box-boards/page/86/#comment-190270](https://forum.armbian.com/topic/34923-csc-armbian-for-rk322x-tv-box-boards/page/86/#comment-190270)

The build framework still shows eos distributions (currently it shows buster, focal, kinetic, lunar as eos) but if you choose them, the build fails as the build system tries to pull things from apt.armbian.com for the distribution which no longer exists.

So I would suggest either filtering out the eos distributions from the build framework ui, or remove the eos distributions from the config/distributions directory completely. Or I guess we could also restore the eos distribution content on apt.armbian.com.

Looking for feedback on which way this should be fixed.

### How to reproduce?

performed a build for focal

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-05-09:
I'd vote for removing eos userspaces from framework, at least disable the code necessary to show them even in `EXPERT` mode.
Restoring apt repo for those is possible too but why should we allow resources to be wasted?

igorpecovnik, 2024-05-10:
Now all are present, but EOS are not updating. Errors should be gone on all.

closed 2024-05-10 by igorpecovnik