armbian/build #6568: Fail at check_loop_device: device node doesn't exist and `$LOOP=`

JohnTheCoolingFan, 2024-05-09:
### What happened?

*I did confirm this problem on the main branch*

~~Trying to build armbian on gentoo with docker. Targeting bigtreetech CB2 from https://github.com/bigtreetech/build/tree/CB2~~

~~The fork isn't far behind the main branch, at least in terms of the compile script.~~

The problem seems to be that the script is looking for a loop device, but there aren't any that match a pattern, yet the loop exits on the first iteration with a false positive and empty `LOOP` variable.

paste.armbian.com/log returns a 500 status code, putting the log here: 
[log-build-3ebc61fd-016c-4aa1-9a41-00194251ff89.log.ans.txt](https://github.com/armbian/build/files/15264700/log-build-3ebc61fd-016c-4aa1-9a41-00194251ff89.log.ans.txt)

Also, the log with docker prefix, because the script decided to use docker automatically: 
[log-docker-3ebc61fd-016c-4aa1-9a41-00194251ff89.log.ans.txt](https://github.com/armbian/build/files/15264734/log-docker-3ebc61fd-016c-4aa1-9a41-00194251ff89.log.ans.txt)

### How to reproduce?

(May get updated)

1. Checkout the repo
2. Run `./compile.sh`
3. Make all the selections and wait  until the failing point

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

{message:Error adding document.}

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnTheCoolingFan, 2024-05-11:
It looks like the script needs to call `sudo losetup -f` at least once to make sure there is an available device existing. On my machine, there are no devices that match the pattern of `/dev/loopX[Y]`, just `/dev/loop-control`. After calling the aforementioned command `/dev/loop0` appeared. After doing that, I managed to build an image.

JohnTheCoolingFan, 2024-05-11:
mknod does get called in check_loop_device, but only with a "found" device, not  for creating a missing device. I am putting together a fix for this and will make a PR.

JohnTheCoolingFan, 2024-05-23:
Yes, the commit got reverted until a better solution is made. And thanks for reporting that this is also the case on Fedora, it's probably much easier to test with for others than gentoo

tjhexf, 2024-06-04:
Happening to me under Fedora, any updates?

tdleiyao, 2024-06-09:
manjaro seems to have the same problem

JohnTheCoolingFan, 2024-06-09:
My previous attempted solution was calling `sudo losetup -f` in the script always, but it's problematic because you can't always expect the user to authorize sudo use. It needs to be called on the host and not inside of docker. Sure, you can do this manually, but the script exists for automating it.
So I propose this mechanism: check for `/dev/loop0` to exist, if it's not there, prompt the user whether to call the sudo command from the script, proceed anyway or possibly have the user running the command manually, or abort the process.

rpardini, 2024-06-09:
@JohnTheCoolingFan you can check if inside Docker (where it is always root) or not and adapt. What we can't have is it calling losetup in the outer host in the re-launching scenario (consider armbian/build can run the outer host on MacOS/Darwin and the relaunched in Linux on a VM, as is the case with developers using Intel and Apple Silicon Macs with Docker Desktop / Rancher Desktop / Podman Desktop etc).

JohnTheCoolingFan, 2024-06-10:
My current local fix for this is a script in `userpatches/extensions/losetup.sh`:
```bash
function add_host_dependencies__losetup(){
	if [[ ! -e /dev/loop0 ]]; then
		local sudo_prefix="" && is_root_or_sudo_prefix sudo_prefix
		${sudo_prefix} losetup -f
	fi
}
```

PR #6927 (2024-07-11): Add MKNOD capability to docker container
# Description

Added arguments `--cap-add MKNOD` to `docker run` command to allow `losetup -f` to create the loop device node.
Fixes #6568

[GitHub issue](https://github.com/armbian/build/labels/Task%2FTo-Do) reference: https://github.com/armbian/build/issues/6568
[Jira](https://armbian.atlassian.net/jira) reference number [AR-2132]

# How Has This Been Tested?

Running `./compile.sh build BOARD=odroidm1 BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no KERNEL_GIT=shallow RELEASE=bookworm` (auto defaults to build in docker)

# Checkli [... cut]
closed 2024-07-11 by JohnTheCoolingFan