armbian/build #6587: Ethernet on Orange Pi 3B not working properly

umifly, 2024-05-12:
### What happened?

After flashing the latest Armbian Jammy firmware onto Orange Pi 3B, the Ethernet port does not function properly, and the indicator light only flashes green. The router displays that there’s no connection.

### How to reproduce?

After the Armbian system boots up

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-05-12:
From what I've read at OPis chat it seems like this board is broken by design and will receive a major hardware revision

https://x.na4.eu/KOTu3/QATiHOJO88.jpg

tl;dr: looks like this cannot be fixed by software. 
Suggesting to put board to EOS status in general since I assume the new revision will have the same naming which then, as usual for xunlong, creates additional confusion due to their naming "convention".

EvilOlaf, 2024-05-12:
@amazingfate @rpardini and opinion on this?

amazingfate, 2024-05-12:
I've beening using wireless for a long time and didn't notice issue with ethernet. I will check it later.

igorpecovnik, 2024-05-12:
closing as its not framework problem and community supported target. if its broken, eos it right-away to save our precious time.

closed 2024-05-12 by igorpecovnik

rpardini, 2024-05-12:
I've used ethernet no problem on mine for a long time. I sincerely don't care if it is eos or not, but I will need to keep it building. Having it as eos only makes life harder, since it is no longer included in default inventory.

umifly, 2024-05-12:
I was able to use Ethernet normally on a previously compiled version of the Armbian 23.11 firmware, so I'm unsure whether it's a hardware issue or a problem with the current version of the system.

rpardini, 2024-05-12:
I've the opi3b on a full PXE-based rig, so yeah the eth works (incl stable MAC addr) also in u-boot and iPXE, not only Linux.

I propose to demote all opi boards to EOS (since same "grounds" apply), or bring this one back (to literally anything that's not `eos`). If a different hw revision shows up (they need to fix pcie power) we can always do an orangepi3bv2 or whatever. It's not like they've not already 10 different opi5's.

igorpecovnik, 2024-05-12:
If CSC board is breaking the process or they are reported as not working, we can remove them to save some time. To us and people that might still download and hope they work. Few weeks ago someone told me that we are almost criminals, because we don't write on download pages which functions works and which not ... I told him that this is not possible, every code change can toggle some function working / not working and we don't know. Expensive automation might be able to provide a fraction of that ... no he wanted that we update information about a feature he is interested in ... Users perception will remain this way. They don't learn, they don't explore, they don't reason. They have demands, some high as hell. And we have nothing to serve this.

No decision here is correct.

Community support is good solution in general and until its not abused. This was abused here as this is build system ticketing, where we don't accept any HW related ticket, especially not for CSC / TVB targets. "Customers" will always try to disarm us on emotional ground - their pain and our curiosity of being engineers resolving a problems. We have limits. Once someone cross them, might be lost for years, for good :(

If we don't stick together on this, we will only suffer.

> we can always do an orangepi3bv2 or whatever. 

There are hundreds of models out there. We could peacefully skip this board, especially as it was thrown at us like we are a bunch of dogs. Are we? When we lost decisions on what we will spent our time / life on? Some people hate being forced to deal with support questions on hardware they didn't want to have. Unless this is their full-time job and they are paid for doing what they don't want.

Once again, this here is build framework ticketing system. Which we struggle to maintain! Our documentation is terrible, but we worry about 3rd party business gain. Again. And their customers problems.

amazingfate, 2024-05-13:
The ethernet issue should be hardware related since me and rpardini both do not encounter it. If xunlong could exchange the broken boads so that there is no changes at our side, we can just keep the board as csc.

igorpecovnik, 2024-05-13:
We'll get back to this. Its low priority atm.

upsangel, 2024-07-23:
I have sent the eth problem board to the seller and he forwarded to the factory (Xunlong I guess), then got a new version replacement instead of fixing. From what I heard, it's a hardware design fault cannot be fixed, even fixed it will recur later. So the Gen1 Opi3B should be avoided. 

Having said that, there is an easy "fixed": just plug a RTL8152 / 8153 usb eth to it, armbian just works without any hassels. (But not the factory debian image, that debian driver has problem and the usb eth cannot work without some heavy lifting).