armbian/build #6630: Display is broken on OLIMEX Teres-I

Kreyren, 2024-05-18:
### What happened?

On kernels past 6.1 there have been few reworks of the relevant components that cause the display to not get initialized correctly

I fixed that by adding:

```
usbhid
sun4i-drm
sun4i-tcon
sun8i-mixer
sun8i_tcon_top
drm
drm_shmem_helper
drm_kms_helper
drm_dma_helper
drm_display_helper
analogix_anx6345
analogix_dp
```

to `/etc/initramfs-tools/modules` and then ran `update-initramfs -u`

Self-Assigning.. Currently very busy will fix it when i find a time frame for it.

### How to reproduce?

Run the pre-built image on the device

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 13 Trixie

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-05-22:
Its not a problem if you are busy, but non working display is pretty critical problem for laptop ... as there is no time, I can only disable "supported" tag on download until this is not resolved as providing known broken images for supported hardware is bad for reputation.

Kreyren, 2024-05-23:
> Its not a problem if you are busy, but non working display is pretty critical problem for laptop ... as there is no time, I can only disable "supported" tag on download until this is not resolved as providing known broken images for supported hardware is bad for reputation. -- @igorpecovnik (https://github.com/armbian/build/issues/6630#issuecomment-2124796655)

Teres users are different userbase from the generic consumer base and generally know how to manage this problem + are in talks with me on them (all known problems explained in detail on https://linux-sunxi.org/Olimex_Teres-A64#Known_issues),

So keep the supported tag and instead i will add notes about these issues to the armbian page with links until i track down the cause of this issue in armbian (not known to happen on other distros, the display was broken in non-lts kernels until 6.8 so i assume that someone's applying obsolete patches to the armbian kernel atm)

igorpecovnik, 2024-05-23:
> So keep the supported tag

Its better that its community supported until problems are not resolved as those images are auto-created once per week, one desktop, one CLI image. If someone else fixes this upstream it will just start working ... Once you confirm they work, status is changed back and new stable kernel is pushed to repo + stable images are recreated. This is exception that can be made.

>  i will add notes about these issue

You are free to add comments what users need to do to get going, but it might be less work just fixing it. I understand you are busy, everyone is, this is not a critical problem => when you find time.

>  i assume that someone's applying obsolete patches to the armbian kernel atm

Nobody is deliberately doing anything. I don't know what is the status, nobody knows "at Armbian", which is the reason why we have maintainers. They know. You know. I only know for few boards that I am dealing with and things we have on automated testings.

Kreyren, 2024-07-15:
Should be fixed now, tested and the display worked each time for me using the images from armbian download.

Heads-up: @igorpecovnik mark as supported plz

closed 2024-07-15 by Kreyren

Kreyren, 2024-07-15:
FWIW: If the issue shows up again then it will likely have to be fixed in u-boot to do proper display initialization sequence for the display