armbian/build #6632: OLIMEX Teres-I: parameter rwt_enusbss ignored

Kreyren, 2024-05-19:
### What happened?

[  +0,002463] r8723bs: unknown parameter 'rtw_enusbss' ignored

Minor issues, Self-Assigning..

### How to reproduce?

...

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 13 Trixie

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leggewie, 2025-07-05:
still an issue?

igorpecovnik, 2025-07-06:
AFAIK nobody is maintaining this hw  / wifi and this is not a build FW related anyway, closing.

closed 2025-07-06 by igorpecovnik