armbian/build #6655: PCIe Link width degraded to x1 for RockPro64 (probably all RK3399)

chnce, 2024-05-23:
### What happened?

Some time ago I noticed that that the SATA card on my RockPro64 only uses a x1 PCIe link, even though it supports x2.
In order to narrow the issue down, I tested different combinations of hardware (x2 SATA card, x4 NVMe) and software (latest Armbian, Armbian 23.8, manjaro ARM).
On older Armbian builds as well as on Manjaro, both cards run at their designated link width (x2 and x4).

Comparing the patches between Armbian 23.8 and current showed the introduction of the following patch: https://github.com/armbian/build/pull/6057

Building an image for current Armbian with said patch disabled (for which I by the way had to delete the patch, creating an empty file in user_patches did not work), restores the full PCIe with.

This is my related forum post: **https://forum.armbian.com/topic/39370-pcie-width-only-x1-for-a-x4-card/#comment-191184**

### How to reproduce?

1. Use current image 24.2.1 (or daily build) on a RockPro64 (other RK3399 will probably have the same issue)
2. Install PCIe card with more than x1 capability.
3. Check PCIe lane width with `lspci -vv`. It will only be a x1 link.
4. Build Armbian image with `patch/kernel/archive/rockchip64-6.6/rk3399-fix-pci-phy.patch` disabled and flash to the RockPro64
5. Check PCIe lane width with `lspci -vv`. The link width should correspond to the full capability.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-05-23:
Answered in forums.

closed 2024-05-23 by igorpecovnik

walmis, 2024-12-12:
This patch should fix it, previously PHY_LANE_IDLE_OFF was only disabled for the fist lane

```diff
From 77f6dfcb20c2dc6a4a2f5303709c6fa0c7b65f30 Mon Sep 17 00:00:00 2001
From: Valmantas Paliksa <walmis@gmail.com>
Date: Thu, 12 Dec 2024 12:24:33 +0200
Subject: [PATCH] Disable PHY_LANE_IDLE_OFF for each instance of
 rockchip_pcie_phy_power_one

Previously PHY_LANE_IDLE_OFF was only disabled for the first lane
---
 drivers/phy/rockchip/phy-rockchip-pcie.c | 12 ++++++------
 1 file changed, 6 insertions(+), 6 deletions(-)

diff --git a/drivers/phy/rockchip/phy-rockchip-pcie.c b/drivers/phy/rockchip/phy-rockchip-pcie.c
index 8234b83fdd88..240cb27a0e9e 100644
--- a/drivers/phy/rockchip/phy-rockchip-pcie.c
+++ b/drivers/phy/rockchip/phy-rockchip-pcie.c
@@ -167,6 +167,12 @@ static int rockchip_pcie_phy_power_on(struct phy *phy)
 
 	mutex_lock(&rk_phy->pcie_mutex);
 
+	regmap_write(rk_phy->reg_base,
+		     rk_phy->phy_data->pcie_laneoff,
+		     HIWORD_UPDATE(!PHY_LANE_IDLE_OFF,
+				   PHY_LANE_IDLE_MASK,
+				   PHY_LANE_IDLE_A_SHIFT + inst->index));
+
 	if (rk_phy->pwr_cnt++)
 		goto err_out;
 
@@ -181,12 +187,6 @@ static int rockchip_pcie_phy_power_on(struct phy *phy)
 				   PHY_CFG_ADDR_MASK,
 				   PHY_CFG_ADDR_SHIFT));
 
-	regmap_write(rk_phy->reg_base,
-		     rk_phy->phy_data->pcie_laneoff,
-		     HIWORD_UPDATE(!PHY_LANE_IDLE_OFF,
-				   PHY_LANE_IDLE_MASK,
-				   PHY_LANE_IDLE_A_SHIFT + inst->index));
-
 	/*
 	 * No documented timeout value for phy operation below,
 	 * so we make it large enough here. And we use loop-break
-- 
2.34.1
```

igorpecovnik, 2024-12-12:
I think this bug is already fixed. Is it still happening?

walmis, 2024-12-12:
Yes - without the patch, after warm restart of the rockchip_pcie_phy module the pcie lanes other that 0 remain stuck in PHY_LANE_IDLE_OFF state ant the link remains in x1 mode. 

The patch moves the `regmap_write` block before 
```C
if (rk_phy->pwr_cnt++)
 		goto err_out;
```
so it can reenable other lanes other than the first one.

walmis, 2024-12-13:
Done, thanks

https://github.com/armbian/build/pull/7572