armbian/build #6751: mac address bug

ZYQsteve, 2024-06-17:
### What happened?

How to solve the problem of using the same TF card to execute the armbian install command on some devices with the same parameters, which have the same MAC address

### How to reproduce?

run armbian-install on same parameters devices

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-06-17:
This is custom hardware world and every device act differently. You need to start this way https://www.armbian.com/bugs in a section where they are talking about device you have. 

This here is framework bug tracker which doesn't deal with hw level problems.

EvilOlaf, 2024-06-17:
I'd suggest to start with customizing the image itself: https://docs.armbian.com/Developer-Guide_User-Configurations/
Everything beyond -> forums

closed 2024-06-17 by EvilOlaf