armbian/build #6761: rk3588s Edge: aic8800 install fails and halts image creation

CT1IQI, 2024-06-19:
### What happened?

When compiling edge based image for Rock-3C board (rk3588s SoC), kernel  6.10.0-rc4-edge-rockchip-rk3588 (aarch64)
the process stops due to the inclusion of aic8800 drivers and firmware not succeeding (WiFi/BT).
See attached snippet of the log.
[aicc_error.txt](https://github.com/user-attachments/files/15898180/aicc_error.txt)

### How to reproduce?

Working in VM. Doing ./compile.sh for the Rock-3C board.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-06-19:
Lacking full logs.

CT1IQI, 2024-06-19:
additional full logs:
[log-build-0e1dd9a8-4b94-46cf-89e9-4fc801423e7e.log](https://github.com/user-attachments/files/15899399/log-build-0e1dd9a8-4b94-46cf-89e9-4fc801423e7e.log)

EvilOlaf, 2024-06-19:
Lacking build config...

amazingfate, 2024-06-19:
armbian will download latest release from radxa. And now they have not released the latest code including fix for kernel 6.9. We have to wait.
@RadxaYuntian

EvilOlaf, 2024-06-19:
Actually 6.10. Reporter is using edge.

RadxaYuntian, 2024-06-19:
Can someone first test if there is additional issue with 6.10 before we determine on the package release schedule?

RadxaYuntian, 2024-06-19:
Additionally, we are waiting for Aicsemi to release an updated driver to fix 5C's Wi-Fi crashing issue, so it might be worth to just wait until then before releasing a new version. We have validated the test version they provided today, and are just waiting for an official version.

amazingfate, 2024-06-19:
> Can someone first test if there is additional issue with 6.10 before we determine on the package release schedule?

Packages downloaded from github workflow works with 6.10-rc3. I can use wifi on rock5c.

RadxaYuntian, 2024-06-19:
We will include 5C's Wi-Fi fix tomorrow and release a new version.

RadxaYuntian, 2024-06-20:
Just triggered the new release: https://github.com/radxa-pkg/aic8800/commit/8ad987876a6220fffb40a755e0454250ce2c5fdd. Package should be available shortly.

CT1IQI, 2024-06-20:
Thanks!
There is an adaptation needed to the extensions//radxa-aic8800.sh script because the naming of the files was changed from arch specific to cross-platform 'all'. The script tries to download _arm64 and fails because it is now _all for the various aic8800 files.

CT1IQI, 2024-06-20:
after adaptation of script compiled and loaded on Rock 5C board.
At  [4.120996] dmesg shows a problem while loading firmware for bluetooth. See attached log.
[aic8800_log.txt](https://github.com/user-attachments/files/15911440/aic8800_log.txt)
(kernel 6.10.0-rc4)

RadxaYuntian, 2024-06-20:
Can you test 6.10.0-rc3 as @amazingfate has tested this version, sans the Wi-Fi crash fix.

amazingfate, 2024-06-20:
I have not checked bluetooth with 6.10-rc3. Maybe it has the same firmware loading issue.

RadxaYuntian, 2024-06-20:
@CodeChenL please check the bluetooth driver on 5C with our 6.1 kernel later.

EvilOlaf, 2024-06-20:
Building is now fine. Therefore reporters initial issue has been resolved.

https://paste.armbian.com/ifacoyicir

closed 2024-06-20 by EvilOlaf