armbian/build #6762: Bookworm no longer has a Release file

randoentity, 2024-06-19:
### What happened?

https://armbian.site-meganet.com/apt/dists/bookworm/ is empty while the other point release directories are not.
Is it building or has it been deprecated? Is there an ongoing security threat? I'm not finding anything online.

### How to reproduce?

See https://armbian.site-meganet.com/apt/dists/bookworm/

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-06-19:
Known issue with our apt repo. It is being worked on.

randoentity, 2024-06-19:
@EvilOlaf thanks for the information. I'm new here. Is there a status page or RSS feed or IRC channel for these kind of things? OTOH the directory being empty is quite a tell something is up as well.

EvilOlaf, 2024-06-19:
Here are a few resources if you're interested:
https://status.armbian.com/
https://docs.armbian.com/Community_IRC/

randoentity, 2024-06-19:
Resolved, as far as I can tell!

closed 2024-06-19 by randoentity