armbian/build #6790: Could not find package filename for 'base-files'

The-going, 2024-06-24:
### What happened?

Make a 'Noble' image on Ubuntu 24.04 VM
```
./compile.sh RELEASE=noble
....

[🌱] artifact [ armbian-base-files :: armbian-base-files() ]
[🚸] Could not find package filename for 'base-files' in 'https://packages.ubuntu.com/noble/arm64/base-files/download' [ looking for base-files ]
[🚸] Command failed, retrying in 15s [ apt_find_upstream_package_version_and_download_url base-files ]
[🚸] Could not find package filename for 'base-files' in 'https://packages.ubuntu.com/noble/arm64/base-files/download' [ looking for base-files ]
[🚸] Command failed, retrying in 15s [ apt_find_upstream_package_version_and_download_url base-files ]
[🚸] Could not find package filename for 'base-files' in 'https://packages.ubuntu.com/noble/arm64/base-files/download' [ looking for base-files ]
[🚸] Command failed, retrying in 15s [ apt_find_upstream_package_version_and_download_url base-files ]
curl: (28) Operation timed out after 10001 milliseconds with 0 bytes received
```
Url: [https://packages.ubuntu.com/noble/arm64/base-files/download](https://packages.ubuntu.com/noble/arm64/base-files/download)
It contains only instructions on how to download the source package correctly.

### How to reproduce?

New install Ubuntu 24.04 on Qemu VM.
```
git clone --depth=1 --branch=main https://github.com/armbian/build armbian
cd armbian
./compile.sh RELEASE=noble
```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-06-24:
We can quickly only implement dirty workaround ... Is it possible that you use Debian user spaces until then?

The-going, 2024-06-24:
> We can quickly only implement dirty workaround ... Is it possible that you use Debian user spaces until then?

Good afternoon, everyone.
Igor, this is not a problem for me.
I just marked it here.

The assembly of the Spacemit family is currently possible only for Noble.

For reference.
In my own build system, my algorithm looks like this:
1) Make the build environment the same as the target OS.
For bookworm arm, packages will be collected in bookworm x86_64.
For 'noble', packages will be collected in 'noble x86_64'.

2) Open access to the source packages in /etc/apt/source.list
3)
```
apt-get update
apt-get source $pkg-name
```

In this case, the host OS can be any Linux.

The-going, 2024-06-24:
> Is the website down again? See #6763

@ColorfulRhino If this method works well, maybe it will be the main one?

I'll check it out now.

igorpecovnik, 2024-06-24:
O.K.

> The assembly of the Spacemit family is currently possible only for Noble.

Debian Sid / Trixie should be ready for riscv ... ?

The-going, 2024-06-24:
> Debian Sid / Trixie should be ready for riscv ... ?

There may be problems with the compiler.
I'll try to add an external compiler.
Everything is in progress.

The-going, 2024-06-24:
> > > Is the website down again? See #6763
> > 
> > 
> > @ColorfulRhino If this method works well, maybe it will be the main one?
> > I'll check it out now.
> 
> There is likely room for improvement, e.g. I don't know how my proposed alternative method behaves in a massive build train when building many images at once. But it worked perfectly fine for me when the website was down for like half a day.

Everything works. Thank you.

The-going, 2024-06-24:
The problem is solved in [#6763](https://github.com/armbian/build/pull/6763)

It's closing

closed 2024-06-24 by The-going

MNLierman, 2025-01-02:
This shit is still not fixed. I have an unresovled dependency and I see no way of fixing it. I've been working on this for hours. It seems that "base-files_25.2.0-trunk.252" does not actually exist but it's required for "armbian-bsp-cli-rpi4b-edge_25.2.0-trunk.252_arm64" and since I cannot install either, my system is now broken and will not upgrade. I tried to go from Stable to Beta/Edge, and I'm pretty good with this stuff. Used to compile my own kernel as a kid like 15 years ago, in order to get things like wifi drivers to work. Given this, I'm surprised that I cannot figure this out. I'm resulting to creating a fake "fake-base-files-25.2.0-truck.252 but I have my doubts this is even going to work. Does no one test these things? Can someone point me to where the deb file is so I can attempt to force install? Google and Bing searches for the base-file package only turns up version 3.0-15.0, How is that possible? Did the package some how jump from 15 to 25? 

Yes I've gone through the upgrade path correctly, yes I've done apt update, upgrade, dist-upgrade, do-release-upgrade, etc for the better part of the past 2 hours, mixed with reboots, more Googling, searching tthe logs for any https paths that will lead me to the file. I'm convinced that at this point base-files version 25 is completely non-existent.

Anyone?

amazingfate, 2025-01-03:
> This shit is still not fixed. I have an unresovled dependency and I see no way of fixing it. I've been working on this for hours. It seems that "base-files_25.2.0-trunk.252" does not actually exist but it's required for "armbian-bsp-cli-rpi4b-edge_25.2.0-trunk.252_arm64" and since I cannot install either, my system is now broken and will not upgrade. I tried to go from Stable to Beta/Edge, and I'm pretty good with this stuff. Used to compile my own kernel as a kid like 15 years ago, in order to get things like wifi drivers to work. Given this, I'm surprised that I cannot figure this out. I'm resulting to creating a "fake-base-files-25.2.0-truck.252" to satisfy the dependency but I have my doubts this is even going to work. Does no one test these things? Can someone point me to where the deb file is so I can attempt to force install? Google and Bing searches for the base-file package only turns up version 3.0-15.0, How is that possible? Did the package some how jump from 15 to 25?
> 
> Yes I've gone through the upgrade path correctly, yes I've done apt update, upgrade, dist-upgrade, do-release-upgrade, etc for the better part of the past 2 hours, mixed with reboots, more Googling, searching tthe logs for any https paths that will lead me to the file. I'm convinced that at this point base-files version 25 is completely non-existent.
> 
> Anyone?

I don't have this issue:
```
$ LC_ALL=C sudo apt install armbian-bsp-cli-rock-5b-edge
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  python3-m2crypto
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  base-files
The following packages will be upgraded:
  armbian-bsp-cli-rock-5b-edge base-files
2 upgraded, 0 newly installed, 0 to remove and 302 not upgraded.
Need to get 501 kB of archives.
After this operation, 9216 B of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 https://fi.mirror.armbian.de/beta noble/noble-utils arm64 base-files arm64 25.2.0-trunk.252-13ubuntu10-noble [68.8 kB]
Get:2 https://netcup.armbian.com/beta noble/main arm64 armbian-bsp-cli-rock-5b-edge arm64 25.2.0-trunk.252 [432 kB]
Fetched 501 kB in 10s (48.4 kB/s)
(Reading database ... 487201 files and directories currently installed.)
Preparing to unpack .../base-files_25.2.0-trunk.252-13ubuntu10-noble_arm64.deb ...
Unpacking base-files (25.2.0-trunk.252-13ubuntu10-noble) over (24.8.3-13ubuntu10-noble) ...
Setting up base-files (25.2.0-trunk.252-13ubuntu10-noble) ...
motd-news.timer is a disabled or a static unit not running, not starting it.
motd-news.service is a disabled or a static unit not running, not starting it.
(Reading database ... 487201 files and directories currently installed.)
Preparing to unpack .../armbian-bsp-cli-rock-5b-edge_25.2.0-trunk.252_arm64.deb ...
Armbian 'armbian-bsp-cli-rock-5b-edge' for '1-PC7e64-V19d9-H2670-B6e44-R89c9': 'preinst' starting.
Armbian 'armbian-bsp-cli-rock-5b-edge' for '1-PC7e64-V19d9-H2670-B6e44-R89c9': 'preinst' finishing.
Unpacking armbian-bsp-cli-rock-5b-edge (25.2.0-trunk.252) over (24.8.3) ...
Armbian 'armbian-bsp-cli-rock-5b-edge' for '1-PC765f-V9dfe-H2670-Bb573-R89c9': 'postrm' starting.
Armbian 'armbian-bsp-cli-rock-5b-edge' for '1-PC765f-V9dfe-H2670-Bb573-R89c9': 'postrm' finishing.
Setting up armbian-bsp-cli-rock-5b-edge (25.2.0-trunk.252) ...
Installing new version of config file /usr/lib/armbian/armbian-apt-updates ...
Armbian 'armbian-bsp-cli-rock-5b-edge' for '1-PC7e64-V19d9-H2670-B6e44-R89c9': 'postinst' starting.
Armbian 'armbian-bsp-cli-rock-5b-edge' for '1-PC7e64-V19d9-H2670-B6e44-R89c9': 'postinst' finishing.
Processing triggers for cracklib-runtime (2.9.6-5.1build2) ...
Processing triggers for plymouth-theme-ubuntu-text (24.004.60-1ubuntu7) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for install-info (7.1-3build2) ...
Processing triggers for plymouth-theme-ubuntu-budgie-text (0.19.0) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.142ubuntu25.2) ...
update-initramfs: Generating /boot/initrd.img-6.13.0-rc5-edge-rockchip64
update-initramfs: Armbian: Converting to u-boot format: /boot/uInitrd-6.13.0-rc5-edge-rockchip64
Image Name:   uInitrd
Created:      Fri Jan  3 10:35:20 2025
Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
Data Size:    24447137 Bytes = 23874.16 KiB = 23.31 MiB
Load Address: 00000000
Entry Point:  00000000
update-initramfs: Armbian: Symlinking /boot/uInitrd-6.13.0-rc5-edge-rockchip64 to /boot/uInitrd
'/boot/uInitrd' -> 'uInitrd-6.13.0-rc5-edge-rockchip64'
update-initramfs: Armbian: done.
Processing triggers for man-db (2.12.0-4build2) ...
```

MNLierman, 2025-01-03:
Since you are able to download `base-files_25.2.0-trunk.252-13ubuntu10-noble_arm64.deb` could you please upload that here? You could also go `apt-cache policy base-files` and see what versions are available to you.? 

It looks like you have arm64, and that was going to be my next possible guess was that maybe it was an architecture issue, but clearly not.

amazingfate, 2025-01-03:
You have provided very less information and we can not reproduce your issue.
Package of rpi4b is installable:
```
$ LC_ALL=C apt install armbian-bsp-cli-rpi4b-edge --dry-run
NOTE: This is only a simulation!
      apt needs root privileges for real execution.
      Keep also in mind that locking is deactivated,
      so don't depend on the relevance to the real current situation!
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  python3-m2crypto
Use 'apt autoremove' to remove it.
The following NEW packages will be installed:
  armbian-bsp-cli-rpi4b-edge
0 upgraded, 1 newly installed, 0 to remove and 302 not upgraded.
Inst armbian-bsp-cli-rpi4b-edge (25.2.0-trunk.252 Armbian:noble [arm64])
Conf armbian-bsp-cli-rpi4b-edge (25.2.0-trunk.252 Armbian:noble [arm64])
```

Packages are from armbian beta repo:
```
$ LC_ALL=C apt-cache policy armbian-bsp-cli-rpi4b-edge
armbian-bsp-cli-rpi4b-edge:
  Installed: (none)
  Candidate: 25.2.0-trunk.252
  Version table:
     25.2.0-trunk.252 500
        500 https://beta.armbian.com noble/main arm64 Packages
     24.11.1 500
        500 https://mirrors.tuna.tsinghua.edu.cn/armbian noble/main arm64 Packages
     24.8.4 500
        500 https://mirrors.tuna.tsinghua.edu.cn/armbian noble/main arm64 Packages
     24.8.3 500
        500 https://mirrors.tuna.tsinghua.edu.cn/armbian noble/main arm64 Packages
     24.5.1 500
        500 https://mirrors.tuna.tsinghua.edu.cn/armbian noble/main arm64 Packages
     24.2.1 500
        500 https://mirrors.tuna.tsinghua.edu.cn/armbian noble/main arm64 Packages
```

igorpecovnik, 2025-01-03:
> so something is wrong with what I'm able to access thought apt.

Bugs exists and they will always be with us ;) Open a ticket and provide all information possible.

> Does no one test these things?

- daily automated (sadly [basic / primitive](https://github.com/armbian/os?tab=readme-ov-file#latest-smoke-tests-results), currently even broken, but still way over [public support share](https://github.com/sponsors/armbian)).
- daily manual by many users and developers

But it is impossible to test everything on every possible hardware targets. With complete absence of funding, there is little we can do. This is custom hardware world, not a copy-paste yet another Linux distributions that any kid can assemble without any need for testing.

MNLierman, 2025-01-07:
> Bugs exists and they will always be with us ;) Open a ticket and provide all information possible.
>
>
> - daily automated (sadly [basic / [...]

Thanks for the reply. I completely understand. What can I expect from free software? I do think I will be sticky with Armbian.

I actually figured out what the issue was, it was super simple, all I had to do was download that package manually and then Force all. I think this happened because of a discrepancy when moving through the update channels to try to get to Oracular from Noble. I followed all the source steps including modifying the apt sources, applying updates and upgrades, rebooting, and applying, until zero upgrades / changes were needed, allowing do-dist-upgrade. 

However, I ran into a new issue I would like to bring to your attention. I ended up in a situation where I thought I could just apply several changes and then reboot at the end. I didn't see any harm in doing this. So using armbian-config, I selected to switch channels to edge, but since there was already a pendingrestart registered with pendingrestart the kernel was unable to be changed again. Or at least it gave an error, in the background, I saw it flash three times, and the config continued on, proceeding to autopurge all instances of the kernel it was supposed to be removing. However, due to the pending reboot, it was unable to actually install the kernel that it was supposed to. So I ended up in a state where it removed the armbian-current kernel and did not install the armbian-edge kernel. It may have even unpacked and repacked initrd with this same change. Now it won't boot. 

I took the SD card out, plugged it into another system, CH rooted into it, and try to roll back changes, try to check the logs, and was unsure exactly what it did. So I tried to install the current kernel which would have been the one on there before it's changes. That didn't work, so I uploaded fresh files from a working image from the /boot dir and that didn't work either. I'm wondering if the actual partition #1 (kernel boot?) was modified by the config tool when it attempted to upgrade to the edge kernel. Because it didn't check for a pending rebooting, it essentially purged all kernels from the OS, and now I can't reverse that. Please incorporate a check in armbian-config which changed if pendingrestart is triggered and warn against making changes in this state. You should recommend that the user reboot so that pendingrestart flag is removed, before they attempt to use armbian-config.

igorpecovnik, 2025-01-07:
>  I think this happened because of a discrepancy when moving through the update channels to try to get to Oracular from Noble. 

This is known recipe for troubles. I think this distro upgrade is not 100% successful procedure, not even on stock Ubuntu or Debian. We don't add much, but base files are certainly a package that can make troubles you experienced. Still, I remain optimistic and am trying to come up [with a solution](https://github.com/armbian/configng/blob/main/tools/modules/system/release_upgrade.sh). (function is disabled in the menu as it doesn't work well).

> So using armbian-config, I selected to switch channels to edge, but since there was already a pendingrestart registered with pendingrestart the kernel was unable to be changed again.

Kernel changing should be quite resilient, but you might hit some corner case that we haven't really though on. It should work:
- if connectivity exists
- if index is updated (i think it also updates it before)
- it does not try to change kernel if it can't download needed packages
- if kernel branch is not defined, if fail over to "current"
- ...

> So I ended up in a state where it removed the armbian-current kernel and did not install the armbian-edge kernel.

This situation should not occur under any circumstances. [Check](https://github.com/armbian/configng/blob/main/tools/modules/system/module_armbian_firmware.sh) if you can see what I don't.

> I'm wondering if the actual partition https://github.com/armbian/build/issues/1 (kernel boot?) was modified by the config tool when it attempted to upgrade to the edge kernel.

No, nothing like that. What you are describing is strange. Hmm. The only special thing is that after kernel is installed, initrd is generated. If that is somehow absent or if content of /boot is somehow deleted ... then this can lead into not booting situation. 

Try to recreate this problem and mark down each step. Currently it doesn't ring any bells how you ended up in this situation. Perhaps its to early hour (drinking 1st coffe) :)