armbian/build #6862: [Bug]: several boot loaders doesn't compile when switching host to Noble

igorpecovnik, 2024-07-02:
### What happened?

[image]

### How to reproduce?

./compile  DOCKER_ARMBIAN_BASE_IMAGE="ubuntu:noble"

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/duyodikako

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

The-going, 2024-07-02:
On ubuntu-22.04 all build target done, but 1 bpi-f3 fail
On ubuntu-24.04 12 target fail

Did I understand correctly?

igorpecovnik, 2024-07-02:
> On ubuntu-22.04 all build target done, but 1 bpi-f3 fail
yes
> On ubuntu-24.04 12 target fail
yes

The-going, 2024-07-02:
Currently, there is an old and bad external compiler mechanism in the build system.
It's bad because everything is loaded, but you only need to download the necessary version for the architecture.
Or use one that already exists with the user. That is, just specify the path to the folder.

If you approve, I will implement it.
The riscv64 compiler can be taken here:
[http://archive.spacemit.com/toolchain/](http://archive.spacemit.com/toolchain/)

https://archive.spacemit.com/toolchain/spacemit-toolchain-linux-glibc-x86_64-v1.0.0.tar.xz
or
https://archive.spacemit.com/toolchain/spacemit-toolchain-linux-glibc-x86_64-v0.3.3.tar.xz

After that, some special cases can be solved using an external compiler from buildroot or another, or self-assembled.

amazingfate, 2024-07-02:
I just fixed uboot of rock5b building with -Werror: https://github.com/radxa/u-boot/pull/81
But there are so many rockchip's vendor uboot sources.

igorpecovnik, 2024-07-03:
There are several others:
https://github.com/armbian/os/actions/runs/9769962830/attempts/5

igorpecovnik, 2024-07-11:
[image]

lanefu, 2024-07-11:
> Yeah I have noticed Indiedroid Nova failing to build, it's using some external vendor U-Boot. @lanefu You may consider also moving to the default family uboot :)
> 
> Next set of fixes: #6877

Thanks for heads up.  Ill try to dig out and tinker with it soon.

rpardini, 2024-07-17:
@ColorfulRhino I guess there is a bug introduced here https://github.com/armbian/build/pull/6819 -- you're reading `uboot_cflags` into `uboot_make_envs` before it is set. Thus all u-boots are being built _without_ the CFLAGS that might tone down some errors to warnings. That explains also the problem @amazingfate fixed upstream (`-Werror=maybe-uninitialized` etc) that was previously forcibly ignored by cflags.

igorpecovnik, 2025-04-30:
Now we need to move to Noble. I will start to disable those that doesn't build.

[image]

igorpecovnik, 2025-08-19:
Some weird devices that are not actively monitored probably still have problems, but those we won't be fixing anyway. I think we can close this down.