armbian/build #6938: [Question]: Disable the smaller workspace at startup

paulgheorghecristian, 2024-07-13:
### Which feature would you like to have?

Hello, how can I disable the feature that, at startup, the workspace is smaller and I have to click ESC on the keyboard or click the workspace to make it fullscreen? Thanks!

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

igorpecovnik, 2024-07-13:
We already provide enough variants: minimal images, which can be upgraded or desktop that you can downgrade during build process. Image customization is possible during build time this way:
https://docs.armbian.com/Developer-Guide_User-Configurations/#user-provided-image-customization-script
or via extensions
https://docs.armbian.com/Developer-Guide_Extensions/ ([examples](https://github.com/armbian/os/tree/main/userpatches/extensions))
Regardless of hardware and variant. You can start with minimal CLI and build whatever you need. If you expect we design a custom image for you, start with [Gold partnership](https://forum.armbian.com/subscriptions/), or dig in, spent your own time time exploring, use [community resources](https://forum.armbian.com/). You are certainly not the only one with such idea.

paulgheorghecristian, 2024-07-13:
Not a bug, I just want fullscreen at start up, without the workspaces being smaller at the start. So disable the top left corner button I guess. I will inspect the links, thanks!

amazingfate, 2024-07-13:
Try this extension: https://extensions.gnome.org/extension/4099/no-overview/

igorpecovnik, 2024-07-13:
Aha, now I get it. Probably some Gnome settings or what AF proposes - extension.

paulgheorghecristian, 2024-07-13:
> Try this extension: https://extensions.gnome.org/extension/4099/no-overview/

Wow, thank you so much, this is exactly what I was looking for :D

closed 2024-07-13 by EvilOlaf