armbian/build #6956: Parted compatibility issue related to Ubuntu Noble

igorpecovnik, 2024-07-17:
Bumping build host changed this
https://github.com/armbian/build/blob/0898df5db4b9108a36bcd504c94190be57b5c2fe/config/sources/families/k3.conf#L86
into a problem.

igorpecovnik, 2024-07-22:
Check also this one @3V3RYONE

3V3RYONE, 2024-07-23:
On it!

igorpecovnik, 2024-07-23:
> Not sure what broke in the latest parted shipped with Nobel

With this we can still wait awhile, perhaps it will be sorted out upstream, its not urgent, as not all other build targets can be build with Noble. I would say prepare a PR that can be merged if this doesn't sort out with parted.

igorpecovnik, 2025-01-03:
We are moving to Noble soon, so this is now getting higher priority.
https://paste.armbian.de/rumamefoza

rpardini, 2025-01-03:
Is this _only_ about `k3`?

igorpecovnik, 2025-01-04:
yes, i think so.

3V3RYONE, 2025-01-04:
This one slipped.. Will take a look in couple days

3V3RYONE, 2025-01-06:
@igorpecovnik Tried a local build with noble filesystem. Seems to be working okay for me... Board = sk-tda4vm

Also, I see some noble pre-built images hosted on [armbian site for tda4vm sk](https://www.armbian.com/texas-instruments-sk-tda4vm/). Is it specific to some board?

3V3RYONE, 2025-01-06:
Nonetheless, I applied @glneo 's patch above and built it for SK-TDA4VM board. Build was successful. Need to do sanity boot testing tomorrow.  @igorpecovnik Could you please check if it solves your build failure? Thanks..

igorpecovnik, 2025-01-06:
@3V3RYONE Test by building with parameter:  DOCKER_ARMBIAN_BASE_IMAGE="ubuntu:noble"

3V3RYONE, 2025-01-07:
Thanks @igorpecovnik ! I was able to reproduce it, will post PR soon.

PR #7677 (2025-01-08): K3: Use sfdisk instead of parted for setting flag in boot partition
The latest 'parted' package that is shipped with Ubuntu noble does not allow to set the 'lba' flag on MBR/DOS partitions and fails with an error. Use 'sfdisk' instead for the same.

Fixes #6956 

[GitHub issue](https://github.com/armbian/build/issues/6956)
[Jira issue](https://armbian.atlassian.net/browse/AR-2420)

# How Has This Been Tested?
- [ ] Build and Boot tested on SK-AM62B board.
- [ ] Build and Boot tested on SK-TDA4VM board.

# Checklist:
- [ ] My code follows the style guidelines of this project
- [ ] I have perfor [... cut]
closed 2025-01-08 by igorpecovnik