armbian/build #6966: [Bug]: chroot: failed to run command ‘/bin/bash’: Exec format error on some runners

igorpecovnik, 2024-07-18:
### What happened?

Build fails on GitHub hosted runners.

Current runner version: '2.317.0'
Operating System
  Ubuntu 22.04.4
  LTS

### How to reproduce?

Build on GH actions while using default ubuntu-latest

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.de/izedacisah

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-07-19:
I don't know if this was working before. Is this qemu related? I found this from a quick search: https://github.com/docker/setup-qemu-action

adeepn, 2024-07-19:
check for installed package qemu-user-static or use https://github.com/docker/setup-qemu-action

rpardini, 2024-07-20:
> ### On which host OS are you running the build script and observing this problem?
> Ubuntu 22.04 Jammy
> 
> ### Relevant log URL
> [paste.armbian.de/izedacisah](https://paste.armbian.de/izedacisah)

Logs indicate Noble, not Jammy.

igorpecovnik, 2024-07-20:
Currently not all artifacts can be assembled with Noble, so I am manually switching here and there. I was hoping this change won't be that troublesome so doing this for a week or so ain't problematic, ...

So, Noble docker image on Jammy build host.

igorpecovnik, 2025-08-25:
I don't recall too see them lately. Let's close it down.