armbian/build #6967: [Feature Request]: Add support for Orange Pi 5 Max

tyrius02, 2024-07-18:
### Which feature would you like to have?

The Orange Pi 5 Max is a very recent addition to the OPi5 lineup. It seems to be a bit of a combination of the Plus and the Pro. Here are the differences, AFAICT, between the Pro and the Max.

| Feature | Pro | Max |
| - | --- | --- |
| CPU | RK3588S | RK3588 |
| Memory | LPDDR4 | LPDDR5 |
| USB | 1x USB 2.0 header<br>1x USB 3.1Type A + 1x USB 2.0 Type A<br>2x USB 2.0 Type A<br>1x USB 3 Type C (power)| 2x USB 3.0 Type A<br>2x USB 2.0<br>1x USB 3 Type C (power)|
| Ethernet | 1x GbE | 1x 2.5GbE |
| WiFi | Wi-Fi 5+BT 5.0/BLE (AP6256) | Wi-Fi 6E + BT 5.3/BLE (AP6611)|

The max apparently has three camera connectors along with a PCIe connector while the pro only has two camera connectors. I'm not sure what this is all about.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

tyrius02, 2024-07-18:
I can do some of the lifting if needed, shouldn't be too hard to create some sort of Frankenstein from the pro and plus device trees.

I don't know what to do with the PCIe expansion nor the various camera connectors.

EvilOlaf, 2024-07-19:
Armbian will not consider this request and maintenance without funding.
However anyone from the community is free to donate their time adding support for new hardware.

EvilOlaf, 2024-09-17:
Yeah, I guess so.

closed 2024-09-17 by EvilOlaf

DiegoBM, 2024-09-17:
Shouldn't it be closed as not planned/won't be done? Closing as completed is misleading

SuperKali, 2024-09-17:
> Shouldn't it be closed as not planned/won't be done? Closing as completed is misleading

No because it was implemented and the support was added from community

DiegoBM, 2024-09-17:
Where is it? I can't see it on the main page?

EvilOlaf, 2024-09-17:
I think closing issues on GH isn't automatically synced to Jira...which is kind of annoying but mostly a cosmetically issue.

SuperKali, 2024-09-17:
> I think closing issues on GH isn't automatically synced to Jira...which is kind of annoying but mostly a cosmetically issue.

I closed it too on Jira

alarik84, 2024-11-23:
I was able to install Armbian Server edition v6.1 on my Orange pi 5 max. In one of Armbian images I found file for Orange pi 5 Max. 
[rk3588-orangepi-5-max.zip](https://github.com/user-attachments/files/17877733/rk3588-orangepi-5-max.zip)
I downloaded image for Server images with Armbian Linux v6.1: [Debian 12 (Bookworm)](https://dl.armbian.com/orangepi5-plus/Bookworm_vendor_server)
Then I mounted it in my linux file stystem.
Mounting procedure can be found here: https://www.linuxquestions.org/questions/linux-general-1/how-to-mount-img-file-882386/

Then added attached file into /boot/dtb-6.1.75-vendor-rk35xx\rockchip\ directory.
Please remember to change ownership and permissons to match the other files. Then modifue /boot/armbianEnv.txt:
fdtfile=rockchip/rk3588-orangepi-5-max.dtb
When you save the file please ensure you changed permissions to this file so they match other files in directory.
Then unmount the image and you're ready to install.

It worked for me. I don't need wifi or bluetooth so didn't check that. The only thing I found is that top USB 3.0 port doesn't work. Rest seems to be ggod: got it installed on NVME and have 2 USB ports in use (one USB 2.0 and one USB 3.0).

Good luck.