armbian/build #6968: [Bug]: pinebookpro build image can't boot from towboot

reukiodo, 2024-07-18:
### What happened?

Both the official Armbian as well as any build this creates results in an invalid checksum of the initramfs which towboot cannot boot.

```
Moving Image from 0x2080000 to 0x2200000, end=3f40000
## Loading init Ramdisk from Legacy Image at 08000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    21077355 Bytes = 20.1 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... Bad Data CRC
Ramdisk image is corrupt or invalid
SCRIPT FAILED: continuing...
BootOrder not defined
EFI boot manager: Cannot load any image
SD Boot failed.
```

Removing towboot from SPI results in the included uboot loading (and not validating checksum?) and booting from the image.

See: https://github.com/Tow-Boot/Tow-Boot/issues/319

### How to reproduce?

originally just `./compile.sh` which succeeded to build while failing to boot from towboot but later with `./compile.sh SHARE_LOG=yes` which fails to build at all

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/iveduvemom

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-07-19:
This bug tracker does not accept bugs which are not related to the build framework and the _building_ process. Please refer to armbian.com/bugs

closed 2024-07-19 by EvilOlaf

reukiodo, 2024-07-22:
Ah, so generating a valid checksum is not part of the build process?

If the checksum is not generated at build time, when is it generated?