armbian/build #6993: [Bug]:/usr/bin/docker: Argument list too long

liu2-3zhi, 2024-07-25:
### What happened?

🌱] Socket /run/user/1000/keyring/ssh [ SSH agent forwarding into Docker ]
[🌱] Launching Docker shell [ docker-shell ]
/home/zelly/桌面/armbian-build/build/lib/functions/cli/cli-docker.sh: 行 80: /usr/bin/docker: 参数列表过长
[💥] Error 126 occurred in main shell [ at /home/zelly/桌面/armbian-build/build/lib/functions/cli/cli-docker.sh:80
                 cli_docker_run() --> lib/functions/cli/cli-docker.sh:80
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> ./compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]

### How to reproduce?

Use Docker to Build

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/pikunapaka

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-07-25:
> Use Docker to Build

Please provide full command.

liu2-3zhi, 2024-07-25:
What needs to be done now is to shorten the length of the docker run command. Perhaps you can consider using docker compose.

igorpecovnik, 2024-07-25:
Docker is used automatically and docker-shell is, IIRC, deprecated.

liu2-3zhi, 2024-07-25:
I also got the same error when using docker automatic mode.
The instructions are as follows (please note that I added a new configuration file baicai-h6-pro, but the key to the problem is that the docker run command is too long)

./compile.sh docker-shell BOARD=baicai-h6-pro BRANCH=current BUILD_DESKTOP=yes BUILD_MINIMAL=no DESKTOP_APPGROUPS_SELECTED='browsers chat desktop_tools editors email internet multimedia office programming remote_desktop' DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base KERNEL_CONFIGURE=yes RELEASE=noble

liu2-3zhi, 2024-07-25:
I also got the same error when using docker automatic mode.
The instructions are as follows (please note that I added a new configuration file baicai-h6-pro, but the key to the problem is that the docker run command is too long)

./compile.sh docker BOARD=baicai-h6-pro BRANCH=current BUILD_DESKTOP=yes BUILD_MINIMAL=no DESKTOP_APPGROUPS_SELECTED='browsers chat desktop_tools editors email internet multimedia office programming remote_desktop' DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base KERNEL_CONFIGURE=yes RELEASE=noble

igorpecovnik, 2024-07-25:
Try this way - (without docker command)

./compile.sh BOARD=baicai-h6-pro BRANCH=current BUILD_DESKTOP=yes BUILD_MINIMAL=no DESKTOP_APPGROUPS_SELECTED='browsers chat desktop_tools editors email internet multimedia office programming remote_desktop' DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base KERNEL_CONFIGURE=yes RELEASE=noble

liu2-3zhi, 2024-07-25:
I'm very sorry, I'm using Ubuntu 24.04 which is not supported, I can only compile using docker.

The-going, 2024-07-25:
> I'm very sorry, I'm using Ubuntu 24.04 which is not supported, I can only compile using docker.

I use openSUSE and run in QEMU\KVM+ Virtual Machine Manager

liu2-3zhi, 2024-07-25:
I tested it and found that it can be compiled successfully without installing Docker in Ubuntu 22.04. But it is undeniable that there is still a problem with the Docker run command.

liu2-3zhi, 2024-07-25:
If I compile with docker, I just get /usr/bin/docker: Argument list too long. what currently need is to split the long docker run command to solve the problem.

igorpecovnik, 2024-07-25:
Does this works for you:

`./compile.sh build BOARD=orangepioneplus BRANCH=current BUILD_DESKTOP=yes BUILD_MINIMAL=no DESKTOP_APPGROUPS_SELECTED='browsers chat desktop_tools editors email internet multimedia office programming remote_desktop' DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base KERNEL_CONFIGURE=yes RELEASE=noble`

liu2-3zhi, 2024-07-25:
It seems that the instructions do not add docker. As long as docker is not used, the compilation can be completed on Ubuntu 22.04.

igorpecovnik, 2024-07-25:
> As long as docker is not used

Can you provide logs for this build?

liu2-3zhi, 2024-07-25:
I can't test it yet, but I seem to have found a log file that compiled without docker on ubuntu 24.04. I just want to report that the docker instructions are too long, not that I can't compile at all.
[log-build-28f0dd7f-cae9-49fb-a243-d768499914b2.log](https://github.com/user-attachments/files/16379071/log-build-28f0dd7f-cae9-49fb-a243-d768499914b2.log)

Log when using docker compilation

https://paste.armbian.com/pikunapaka

SuperKali, 2024-09-06:
i can't reproduce your problem, on my host work correctly

PR #7644 (2025-01-03): cli-docker: avoid `argument list too long` errors when too many git changes present; fixes #6993
#### cli-docker: avoid `argument list too long` errors when too many git changes present; fixes #6993

- cli-docker: avoid `argument list too long` errors when too many git changes present; fixes #6993
commit a408e0cb (2025-01-03): cli-docker: avoid `argument list too long` errors when too many git changes present; fixes #6993
closed 2025-01-03 by igorpecovnik