armbian/build #6999: [Bug]: Compilation error encountered with board-rockpis-0001-arm64-dts.patch？？

Lemon1151, 2024-07-27:
### What happened?

Compilation error encountered with board-rockpis-0001-arm64-dts.patch
[image]

### How to reproduce?

v24.05 Compilation error encountered with board-rockpis-0001-arm64-dts.patch

### Branch

other (only relevant in corner cases)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-07-27:
full logs missing.

Lemon1151, 2024-07-28:
> full logs missing.

logs
[log-build-546e4cb0-8cdc-4d42-941d-c3b083618b24.log](https://github.com/user-attachments/files/16401980/log-build-546e4cb0-8cdc-4d42-941d-c3b083618b24.log)
[summary-build-546e4cb0-8cdc-4d42-941d-c3b083618b24.md](https://github.com/user-attachments/files/16402007/summary-build-546e4cb0-8cdc-4d42-941d-c3b083618b24.md)
[log-build-546e4cb0-8cdc-4d42-941d-c3b083618b24.log.ans.log](https://github.com/user-attachments/files/16402008/log-build-546e4cb0-8cdc-4d42-941d-c3b083618b24.log.ans.log)

Lemon1151, 2024-07-28:
There's another build error
[image]
logs
[log-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.log](https://github.com/user-attachments/files/16402051/log-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.log)
[log-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.log.ans.log](https://github.com/user-attachments/files/16402052/log-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.log.ans.log)
[summary-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.md](https://github.com/user-attachments/files/16402053/summary-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.md)

Lemon1151, 2024-07-28:
> There's another build error ![QQ_1722134776600](https://private-user-images.githubusercontent.com/149338935/352786778-8a72629b-cda9-408d-ae5b-02501532b5a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIxMzg2MzksIm5iZiI6MTcyMjEzODMzOSwicGF0aCI6Ii8xNDkzMzg5MzUvMzUyNzg2Nzc4LThhNzI2MjliLWNkYTktNDA4ZC1hZTViLTAyNTAxNTMyYjVhOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyOFQwMzQ1MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MTlmOGQ3MmM1MGUyN2Q4NDBmODA0ZDg4YzlkMzg0MDcyMzRkMWY4ZGNjNzZlMDRjNGI4ODgwYzhiMzJiZTdhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.7Y25S9fxd9wmqhYhyT6LHOGtNWKuTvnfbwtMqW3qGkc) logs [log-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.log](https://github.com/user-attachments/files/16402051/log-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.log) [log-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.log.ans.log](https://github.com/user-attachments/files/16402052/log-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.log.ans.log) [summary-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.md](https://github.com/user-attachments/files/16402053/summary-build-5a4e3a3c-cce4-4bb4-a715-255e9b414f61.md)

The build for Debian can complete normally, but the build for Ubuntu encounters this error.Why is that?

EvilOlaf, 2024-07-28:
`KERNEL_CONFIGURE` switch is deprecated. The correct way is to use `kernel-config` build command and then, after moving the config to `userpatches` directory, use `kernel` command.

Have you retried with `main` branch?

Lemon1151, 2024-07-29:
> `KERNEL_CONFIGURE` switch is deprecated. The correct way is to use build command and then, after moving the config to directory, use command.`kernel-config``userpatches``kernel`
> 
> Have you retried with branch?`main`

Specifically, how should I proceed? The main branch can be built normally, but building Armbian based on Ubuntu for version 24.05 fails, while building Armbian based on Debian for the same version works.

igorpecovnik, 2024-07-29:
> Specifically, how should I proceed? 

If we would have few full time engineers that you would [support](https://github.com/sponsors/armbian), we would maintain 24.05 branch and many other things ... We try to keep previous branches in frozen state, but it seems we are not fully there yet. I hope next 24.08 will stay longer (forever) in good shape.

You have those options:

1. Fix problems with Ubuntu in 24.05 so it will build again
2. Use Debian
3. Use main branch.

Lemon1151, 2024-07-29:
> If we would have few full time engineers that you would [support](https://github.com/sponsors/armbian), we would maintain 24.05 branch and many other things ... We try to keep previous branches in [...]

This is main：
[image]

This is v24.05:
[image]

The error in building debootstrap for v24.05 is caused by the latest version of debootstrap_1.0.137ubuntu2, which was obtained from the "branch:ubuntu/devel".

Lemon1151, 2024-07-29:
It is recommended to fix a stable version of debootstrap instead of obtaining the latest version from the ubuntu/devel branch？

igorpecovnik, 2024-07-29:
> It is recommended to fix a stable version of debootstrap instead of obtaining the latest version from the ubuntu/devel branch？

I have added this https://github.com/armbian/build/commit/9d187dffebd95e7e0a802bc9c2b5f7fb71e3760e to 24.08 branch.

closed 2024-07-29 by igorpecovnik