armbian/build #7002: Armbian & Khadas are rewarding contributors

igorpecovnik, 2024-07-28:
### What happened?

Armbian’s developer community is the backbone of our success, dedicated to [maintaining and updating a wide range of devices](https://www.armbian.com/download). To ensure our long-term sustainability, we need ongoing contributions that support our core infrastructure, including [the build framework](https://github.com/armbian/build), [OS components](https://github.com/armbian/os), and more.

**Introducing the Armbian Contribution Reward Program:**

We are thrilled to launch this initiative to recognize and reward the incredible contributors who help keep Armbian running smoothly. This program not only celebrates your hard work but also clarifies the roles of hardware manufacturers and the larger community in maintaining software support.

**How to get to the pool? Who can participate?**

Pull requests authors that will complete their work awarded with a special tag  GIVEAWAY   
Active community members [that will be nominated](https://forum.armbian.com/topic/42770-nomination-armbian-khadas-are-rewarding-foss-contributors/) by anyone and approved by [at least one moderator](https://forum.armbian.com/members/)
Regulars and first time contributors are both welcome!
You qualify by contributing to:

- [Armbian build framework improvements, bug fixes, completing stalled work,](https://github.com/armbian/build)
- [Enhancements to Armbian documentation,](https://docs.armbian.com/) [adding missing build framework commands](https://docs.armbian.com/Developer-Guide_Build-Options/)
- [Contributions to Armbian build automation components](https://github.com/armbian/os)
- Active participation in [Armbian community](https://forum.armbian.com/) management (e.g., forum management)
- You can find more ideas in our [forums](https://forum.armbian.com/) and [contribution pages](https://www.armbian.com/participate/).

On October 15th, we’ll hold a random draft to select a winner from the pool.

Rewards:

One lucky winner will receive premium and one standard Khadas Mind portable workstation, both with dock, generously donated by Khadas.

[image]

**How to Get Involved:**

Dive into any of the eligible areas and make a significant impact on Armbian. Your contributions not only support our community but also give you a chance to win an amazing reward.

For more details about the reward, visit [Khadas](https://www.khadas.com/product-page/mind) product page!

Let’s join forces to keep Armbian thriving and growing!

### How to reproduce?

H

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [X] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-10-16:
We are going to pull a random draft on upcoming meeting:
https://forum.armbian.com/events/event/50-qa/ 

Who got to the pool?

(part 1, divided into two parts due to GitHub limitations)

@adeepn
@AGM1968
@alex3d
@alexl83
@amazingfate
@andyshrk
@benhoff
@belegdol
@brentr
@Cellularhacker
@chainsx
@chraac
@ColorfulRhino
Didier
@efectn
@EvilOlaf
@ginkage
@Heisath
@hqnicolas
@H-s-c
@hzyitc
@iav
@IsMrX
@jernej
@JMCC
@joekhoobyar
@JohnTheCoolingFan
@Joshua-Riek
@kernelzru
@krachlatte
@Kreyren

igorpecovnik, 2024-10-16:
Part2

@lanefu
@janprunk
@ManoftheSea
@mattx433
MichaelRobinson
@monkaBlyat
@NicoD-SBC
ning
@PanderMusubi
@paolosabatino
@prahal
@pyavitz
@rpardini
@schwar3kat
@SeeleVolleri
@sfx2000
@sicXnull
@SteeManMI
@stephengraf
@SuperKali
@Tearran
@The-going
@TheLinuxBug
@TheSnowfield
@Tonymac32
@Tq-schmiedel
@TRSx80
@tystuyfzand

Kreyren, 2024-10-17:
I don't want to participate in the giveaway as i don't care for proprietary hardware as it's useless brick to me. Thanks for remembering me though

Kreyren, 2024-10-18:
> Writed from an iphone connected through a Huawei wifi router over a starlink connection using an AT&T Submarine Systems -- @hqnicolas (https://github.com/armbian/build/issues/7002#issuecomment-2421074927)

Written by hardware engineer and hackerspace member from heavily modded [OLIMEX Teres-1](https://github.com/OLIMEX/DIY-LAPTOP) managed via [NiXium](https://github.com/NiXium-org/NiXium) through [Quactel EG25-G](https://www.quectel.com/product/lte-eg25-g) running an [open-source firmware](https://hackaday.com/2022/07/12/open-firmware-for-pinephone-lte-modem-whats-up-with-that) which is partially routed through [pilsfree](https://www.pilsfree.net) infrastructure.. :p

If Khadas starts to make OSHW products then let me know i would be happy to help with kernel development and downstream maintenance.

closed 2024-10-18 by igorpecovnik

Kreyren, 2024-10-19:
> Open source hardware is a concept. Like democracy, freedom, laziness, ... just a buzzword, sales label. -- @igorpecovnik (https://github.com/armbian/build/issues/7002#issuecomment-2422004750)

> You would only offer help to open source developers community when all the hard work is either done and/or hardware obsolete? ;) -- @igorpecovnik (https://github.com/armbian/build/issues/7002#issuecomment-2422004750)

Rather critically important functionality that you need to adjust your business plan to be able to maintain economy e.g. what OLIMEX does as i can't easily fix the device e.g.

OlinuXino-A20 that had multiple broken components as it was pulled from a heavily modded commercial 3D printer (Ultimaker <3) that experienced significant overvoltage due to failed experiment and i was able to fix it under 20 minutes through just looking on the kicad project vs. RockPro64 that PINE64 scammed me on by getting me a device that broke after 1 week of collective <10 min runtime that is not covered by any kind of warranty and i can't fix it without exposing myself to potential infringement lawsuit due to them releasing schematics under GPLv3 that does not cover hardware and indirectly threaten me with lawsuits if i were to use those and make gerbers for them which i intend to do anyway once i handle the legal part in more sane way.

.. in addition to being unable to turn the product into actual projects like the OlinuXino-A64 i am constantly making new PCBs for these to have RP3-comparable device for <3 EUR in PCBs (that i can make in hackerspace) and including components.

Releasing SBC without getting the schematics and gerbers just doesn't make sense for anything that i do to justify investing time to that and i consider working on those products as unethical as i am basically just enriching some evident bad actor that doesn't care for open-source and rather uses small investment to make themselves seem like they care in exchange to make ~100 K EUR through getting the additional mainline support which i really hate that armbian does that in addition to keeping products alive that do not follow sane standards (SBBR, EBBR) or are painfully obvious engineering malpractice like some silicons are and would prefer them out of support by armbian as you are making money out of making the life for people who work on open-source more difficult.

(Not implying that the TH1520 is engineering malpractice, i consider that a brave step towards riscv development including all of it's hardware bugs that are very fixable in next iterations that i will be getting to aid further development)

igorpecovnik, 2024-10-20:
# And the winners are:

- Khadas Mind standard @Heisath 
- Khadas Mind premium @iav 

Congratulation!!!

https://github.com/user-attachments/assets/0807c44e-725f-45d9-8977-2b03ec42824b

iav, 2024-10-20:
I am happy!!!
:1st_place_medal: :^)))

igorpecovnik, 2024-10-20:
Shipping data here: https://www.armbian.com/update-data/

Heisath, 2024-10-20:
Hey, many thanks. Thats an awesome looking product!

SeeleVolleri, 2024-10-21:
🥳

TheSnowfield, 2024-10-22:
congrats!