armbian/build #7007: Add board: HiKey970

brycecherry75, 2024-07-29:
### Which feature would you like to have?

When I tried booting from an SD card loaded with Armbian 24.5.1 Noble Gnome on a 96Boards HiKey970, I received a serial console (debug) message stating that a video mode cannot be set.

Note that the HiKey970 console debug port (UART6) USB-serial converter (XR21V1410) cannot transmit when used with generic USB-CDC drivers and therefore requires manufacturer specific drivers in order to function correctly.

The USB ports have a switch as in the HiKey960 which a prebuilt Armbian image is available (the related GPIO assignments in the HiKey960 and HiKey970 are different); also, the hardware functions on the GPIO headers of the HiKey970 are **SPI2/3**, **UART2**, I**2C0/2** (as well as two **ISP I2C buses 0/1**) and additional GPIO is **SPI0** (CAN) and **UART3** (GPS) which I appreciate including drivers for such in a build.

### Funding

- [X] [I am willing to cover expenses](https://www.armbian.com/contact/)

SteeManMI, 2024-07-29:
This issue tracker is for the build framework not for individual board specific issues.  This is a community maintained board, not supported by Armbian.  The proper place to log such issues is in the forums under the Comminity Maintained section.

closed 2024-07-29 by SteeManMI

igorpecovnik, 2024-07-29:
We can try ping Kevin @h-s-c who added this build target. We don't have anyone in the team dealing with this hardware.

h-s-c, 2024-07-29:
@brycecherry75 Did you boot a generic ARM UEFI image with ACPI or did you build a custom image with a hikey970 devicetree? Hikey960/970 were official android developer boards so take a look at kirin/hikey related patches at [https://android-review.linaro.org](https://android-review.linaro.org) and [https://android-review.googlesource.com](https://android-review.googlesource.com) and compare to what reached the upstream kernel. Maybe if you get lucky only minor devicetree changes (like the usb muxing stuff for hikey960) are necessary.