armbian/build #7048: [Bug]: V24.05 Error context msg [ Check MD5 sum of installed packages failed ]

Lemon1151, 2024-08-07:
### What happened?

Error context msg [ Check MD5 sum of installed packages failed ]？

### How to reproduce?

[image]

### Branch

other (only relevant in corner cases)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/log

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Lemon1151, 2024-08-07:
[image]

EvilOlaf, 2024-08-07:
No logs - no issue.

Lemon1151, 2024-08-08:
> No logs - no issue.

logs
[log-build-b69514cb-12a9-4626-9866-caf56aced087.log](https://github.com/user-attachments/files/16536452/log-build-b69514cb-12a9-4626-9866-caf56aced087.log)

igorpecovnik, 2024-08-08:
It looks like previous branch 24.05 is not build-able anymore. Try to use main branch - should work.

Lemon1151, 2024-08-08:
> It looks like previous branch 24.05 is not build-able anymore. Try to use main branch - should work.

Isn't 24.05 a stable version? Why can't the previous version build normally after advancing to the next version?

igorpecovnik, 2024-08-08:
> Why can't the previous version build normally after advancing to the next version?

Perhaps because it costs and [this](https://github.com/sponsors/armbian) only covers coffee I am drinking now while saving you hours or weeks of your precious time.

igorpecovnik, 2024-08-08:
24.08 was frozen today, next stable. Use that.

EvilOlaf, 2024-08-09:
Try wiping your `cache` directory and retry

SuperKali, 2024-08-19:
> Try wiping your `cache` directory and retry

I tried but nothing: https://paste.armbian.com/omicolojig

igorpecovnik, 2024-08-19:
try compiling with FORCE_CHECK_MD5_PACKAGES=no

SuperKali, 2024-08-19:
@igorpecovnik tried to compile by wiping cache and the args: https://paste.armbian.com/arulipahaj but still the same error.

igorpecovnik, 2024-08-19:
Comment # that debsum command then.

SuperKali, 2024-08-19:
ok i will try now.

SuperKali, 2024-08-19:
@igorpecovnik by commenting the debsums command i successfully compiled the OS:

[image]

Log: https://paste.armbian.com/lasibojuwo

leggewie, 2025-08-17:
time to close?

closed 2025-08-20 by Lemon1151