armbian/build #7051: [Bug]:

denprog, 2024-08-08:
### What happened?

I compile in docker with such command line:
`./compile.sh BOARD=orangepizero2 BRANCH=current RELEASE=jammy KERNEL_CONFIGURE=no BUILD_DESKTOP=no BUILD_MINIMAL=yes ALLOW_ROOT=yes ENABLE_EXTENSIONS=prepare_partitions COMPRESS_OUTPUTIMAGE=xz BOOT_LOGO=yes KERNEL_GIT=shallow`

after some time get this error which cannot resolve.

Adding Armbian repository and authentication key [ image-early :: /etc/apt/sources.list.d/armbian.list ]
[🐳|🌱] Disabling Armbian repo [ arm64-jammy :: skip:no when:image-early ]
[🐳|🌱] Applying distribution specific tweaks for [ jammy ]
[🐳|🔨]   Failed to disable unit, unit smartmontools.service does not exist.
[🐳|🔨]   Failed to disable unit, unit smartd.service does not exist.
[🐳|🔨]   Failed to disable unit, unit hostapd.service does not exist.
[🐳|🔨]   Failed to disable unit, unit ondemand.service does not exist.
[🐳|🌿] Installing: /armbian/output/debs/fake-ubuntu-advantage-tools_24.5.0-trunk_all__1-Bfcde-R448a.deb [ fake-ubuntu-advantage-tools_24.5.0-trunk_all__1-Bfcde-R448a.deb ]
[🐳|🔨]   '/armbian/output/debs/fake-ubuntu-advantage-tools_24.5.0-trunk_all__1-Bfcde-R448a.deb' -> '/armbian/.tmp/rootfs-d2fd5518-4186-43f8-92da-04607e82ee44/root/fake-ubuntu-advantage-tools_24.5.0-trunk_all__1-Bfcde-R448a.deb'
[🐳|🔨]   mkdir: created directory '/armbian/.tmp/rootfs-d2fd5518-4186-43f8-92da-04607e82ee44/var/lib/apt/lists'
[🐳|🔨]   Selecting previously unselected package fake-ubuntu-advantage-tools.
[🐳|🔨]   (Reading database ... 18839 files and directories currently installed.)
[🐳|🔨]   Preparing to unpack .../fake-ubuntu-advantage-tools_24.5.0-trunk_all__1-Bfcde-R448a.deb ...
[🐳|🔨]   Unpacking fake-ubuntu-advantage-tools (24.5.0-trunk) ...
[🐳|🔨]   Setting up fake-ubuntu-advantage-tools (24.5.0-trunk) ...
[🐳|🌿] Installing: /armbian/output/debs/jammy/base-files_24.5.0-trunk-12ubuntu4.6-jammy_arm64__1-jammy-1armbian1-Ba4f9-U12ubuntu4.6-R55fa.deb [ base-files_24.5.0-trunk-12ubuntu4.6-jammy_arm64__1-jammy-1armbian1-Ba4f9-U12ubuntu4.6-R55fa.deb ]
[🐳|🔨]   '/armbian/output/debs/jammy/base-files_24.5.0-trunk-12ubuntu4.6-jammy_arm64__1-jammy-1armbian1-Ba4f9-U12ubuntu4.6-R55fa.deb' -> '/armbian/.tmp/rootfs-d2fd5518-4186-43f8-92da-04607e82ee44/root/base-files_24.5.0-trunk-12ubuntu4.6-jammy_arm64__1-jammy-1armbian1-Ba4f9-U12ubuntu4.6-R55fa.deb'
[🐳|🔨]   E: Packages were downgraded and -y was used without --allow-downgrades.
[🐳|💥] Error context msg [ Installation of /root/base-files_24.5.0-trunk-12ubuntu4.6-jammy_arm64__1-jammy-1armbian1-Ba4f9-U12ubuntu4.6-R55fa.deb failed orangepizero2 jammy no sunxi64 ]
[🐳|💥] Error 1 occurred in main shell [ at /armbian/lib/functions/rootfs/apt-install.sh:106
             install_deb_chroot() --> lib/functions/rootfs/apt-install.sh:106
    install_artifact_deb_chroot() --> lib/functions/rootfs/apt-install.sh:121
   install_distribution_specific() --> lib/functions/rootfs/distro-specific.sh:44
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:27
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> ./compile.sh:50
 ]
[🐳|💥] Cleaning up [ please wait for cleanups to finish ]
[🐳|🌿] Unmounting recursively [ SDCARD - be patient ]
[🐳|🌿] Unmounting recursively [ MOUNT - be patient ]
[🐳|🌿] ANSI log file built; inspect it by running: [ less -RS output/logs/log-build-d2fd5518-4186-43f8-92da-04607e82ee44.log.ans ]
[🐳|🌿] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-build-d2fd5518-4186-43f8-92da-04607e82ee44.log.ans https://paste.armbian.com/log ]

### How to reproduce?

./compile.sh docker-shell BOARD=orangepizero2 BRANCH=current RELEASE=jammy KERNEL_CONFIGURE=no BUILD_DESKTOP=no BUILD_MINIMAL=yes BOOT_LOGO=yes

### Branch

other (only relevant in corner cases)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-08-08:
Please switch to v24.08 branch. This problem should be fixed.

denprog, 2024-08-09:
Yes, v24.08 works.

closed 2024-08-09 by denprog