armbian/build #7054: [Bug]: Build fails for desktop on orangepione

stephengraf, 2024-08-09:
### What happened?

🌱] Checking MD5 sum of installed packages [ debsums ]
[🔨]   debsums: changed file /usr/share/doc/base-files/changelog.gz (from base-files package)
[🔨]   debsums: changed file /etc/os-release (from base-files package)
[🔨]   debsums: changed file /etc/issue.net (from base-files package)
[🔨]   debsums: changed file /etc/issue (from base-files package)
[💥] Error 2 occurred in main shell [ at /home/sysadmin/build/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                  chroot_sdcard() --> lib/functions/logging/runners.sh:122
   create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-create.sh:230
        create_new_rootfs_cache() --> lib/functions/rootfs/create-cache.sh:69
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
   artifact_rootfs_build_from_sources() --> lib/functions/artifacts/artifact-rootfs.sh:80
    artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
       obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:280
       build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:392
   get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-cache.sh:83
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💥] Error context msg [ Check MD5 sum of installed packages failed ]

### How to reproduce?

This happens with debian bookworm and ubuntu jammy desktop builds for current in main. Others have not been tested.
It happens when building with my desktop ubuntu 24 and on a KVM ubuntu 22.04 server.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/zimufoqisa

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

stephengraf, 2024-08-09:
I am not sure this is worth fixing. An orangepione with 512MB of memory does not work very well with a desktop image.
However the build should work.
Builds of full server for legacy, current and edge work for orangepione.

EvilOlaf, 2024-08-09:
Probably related to https://github.com/armbian/build/issues/7048 but leaving open since this affects `main`.

igorpecovnik, 2024-08-09:
> I am not sure this is worth fixing. 

Build must work. I could not reproduced on my 22.04 build. (using v24.08 repo, which is identical to main at this point). Will repeat on mail ... need coffee to be able to look into the logs. On first scan, nothing usual ... damn why this base-files makes troubles?

EvilOlaf, 2024-08-09:
Try wiping your `cache` directory and retry.

stephengraf, 2024-08-09:
The build that was reported was the first build job run after the Ubuntu 22.04 vm was created and the repository downloaded.

Tried again with empty cache.

***@***.***:~/build$ sudo rm -r cache/*
***@***.***:~/build$ ls -alh cache
total 8.0K
drwxrwxr-x  2 root     root     4.0K Aug  9 17:10 .
drwxrwxr-x 15 sysadmin sysadmin 4.0K Aug  9 01:22 ..

[🌱] Checking MD5 sum of installed packages [ debsums ]
[🔨]   debsums: changed file /usr/share/doc/base-files/changelog.gz (from base-f              iles package)
[🔨]   debsums: changed file /etc/os-release (from base-files package)
[🔨]   debsums: changed file /etc/issue.net (from base-files package)
[🔨]   debsums: changed file /etc/issue (from base-files package)
[💥] Error 2 occurred in main shell [ at /home/sysadmin/build/lib/functions/logg              ing/runners.sh:211
     run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                   chroot_sdcard() --> lib/functions/logging/runners.sh:122
    create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-cre              ate.sh:230
         create_new_rootfs_cache() --> lib/functions/rootfs/create-cache.sh:69
                 do_with_logging() --> lib/functions/logging/section-logging.sh:8              1
    artifact_rootfs_build_from_sources() --> lib/functions/artifacts/artifact-roo              tfs.sh:80
     artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.s              h:34
        obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.s              h:280
        build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.s              h:392
    get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-ca              che.sh:83
          build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
    full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.s              h:36
           do_with_default_build() --> lib/functions/main/default-build.sh:42
          cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
         armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                  cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                            main() --> compile.sh:50
  ]
[💥] Error context msg [ Check MD5 sum of installed packages failed ]
[💥] Cleaning up [ please wait for cleanups to finish ]

https://paste.armbian.com/vedokeyajo

igorpecovnik, 2024-08-09:
We do have a problem here it seems as there are more reports.

igorpecovnik, 2024-08-09:
Quick fix would be to disable this checking or put it under the switch.

stephengraf, 2024-08-10:
This is not making sense.

I went back installed a new kvm vm and loaded it with ubuntu 22.04lts.
Then I downloaded the 24.05 armbian repository and built an orangepione desktop. It failed to build.

This worked three months ago when I was doing release testing for 24.05.

login as: sysadmin

Welcome to Ubuntu 22.04.4 LTS (GNU/Linux 5.15.0-118-generic x86_64)

Cloning into 'build'...
remote: Enumerating objects: 5695, done.
remote: Counting objects: 100% (5695/5695), done.
remote: Compressing objects: 100% (4519/4519), done.
remote: Total 5695 (delta 1557), reused 3741 (delta 790), pack-reused 0
Receiving objects: 100% (5695/5695), 64.89 MiB | 17.71 MiB/s, done.
Resolving deltas: 100% (1557/1557), done.
Updating files: 100% (6350/6350), done.

https://paste.armbian.com/ipocadidet

EvilOlaf, 2024-08-10:
We are aware of the issue. Seems to occur randomly. Wiping cache sometimes helps. Still investigating.

igorpecovnik, 2024-09-09:
I think this was fixed. I checked and can't reproduce it anymore.

closed 2024-09-09 by igorpecovnik