armbian/build #7055: [Bug]: Can't build legacy branch of kernel for bcm2711

KhazAkar, 2024-08-09:
### What happened?

What happened:
```
[🐳|🔨]   cp: cannot stat '/usr/lib/linux-image-6.1.93-legacy-bcm2711//broadcom/*.dtb': No such file or directory
[🐳|💥] Error 1 occurred in main shell [ at /armbian/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                  chroot_sdcard() --> lib/functions/logging/runners.sh:122
   post_family_tweaks__populate_boot_firmware_directory() --> config/sources/families/bcm2711.conf:184
             post_family_tweaks() --> <extension_magic>:125
          call_extension_method() --> lib/functions/general/extensions.sh:57
   install_distribution_agnostic() --> lib/functions/rootfs/distro-agnostic.sh:393
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:28
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
```
Expectation: It should build, just like `current` branch.

### How to reproduce?

```
./compile.sh RELEASE=jammy BOARD=rpi4b BRANCH=current NETWORKING_STACK=network-manager KERNEL_CONFIGURE=prebuilt BUILD_DESKTOP=no BUILD_MINIMAL=yes EXTERNAL=yes EXTERNAL_NEW=prebuilt INSTALL_HEADERS=yes DISABLE_IPV6=false COMPRESS_OUTPUTIMAGE="sha,xz" KERNEL_GIT=shallow CONSOLE_AUTOLOGIN=no INCLUDE_HOME_DIR=yes
```
Plus, changes mentioned in logs are from this PR: https://github.com/armbian/build/pull/6503 - but since replacing config.txt from customize-image.sh wasn't working with it, I just appended my own lines to this changes - but shouldn't be relevant here.

### Branch

other (only relevant in corner cases)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/qabigucato

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-08-09:
Did you try without modifications?

KhazAkar, 2024-08-09:
@EvilOlaf I'm currently testing `edge` with that change and works. I will stash it, reset and check again on 'legacy' branch.

KhazAkar, 2024-08-09:
Stashed & reset, set to `legacy` branch AAAAAAND... Didn't worked.
https://paste.armbian.com/jalujequzu

igorpecovnik, 2024-08-09:
It works on current and edge, but fails on legacy, right?

KhazAkar, 2024-08-09:
@igorpecovnik exactly that.

igorpecovnik, 2024-08-09:
Why do you need legacy? We don't do much here, so its probably broken / unmaintained upstream.

Those are patches:
https://github.com/armbian/build/tree/main/patch/kernel/archive/bcm2711-6.1
but should be harmless.

Can't you use 6.6.y kernel for your use case?

KhazAkar, 2024-08-09:
I'm trying to get up & running RPi 3 based device, but on 6.6.x kernel I get error -110 for mmc1 device, which is SDIO based wifi card. 6.10 doesn't even work for me there. Since I want to get this specific ones as well as rpi 4, I need to find something which will work on both of them.

igorpecovnik, 2024-08-12:
I was able to build it now.

closed 2024-08-12 by igorpecovnik

Roman3349, 2024-08-14:
I can confirm that I get exactly the same error as KhazAkar when building the legacy branch from the main and v24.08 branches. 

There must be some error when building the dtb packages because the package [https://fi.mirror.armbian.de/apt/pool/main/l/linux-dtb-legacy-bcm2711/linux-dtb-legacy-bcm2711_24.5.1_arm64__6. 1.77-S2101-D440d-Pdbc3-C2812H5c21-HK01ba-Vc222-Bf15a-R448a.deb](https://fi.mirror.armbian.de/apt/pool/main/l/linux-dtb-legacy-bcm2711/linux-dtb-legacy-bcm2711_24.5.1_arm64__6.1.77-S2101-D440d-Pdbc3-C2812H5c21-HK01ba-Vc222-Bf15a-R448a.deb) does not contain any dtb files.

Also packages [https://fi.mirror.armbian.de/apt/pool/main/l/linux-6.1.77/linux-image-legacy-bcm2711_24.5.1_arm64__6.1.77-S2101-D440d-Pdbc3-C2812H5c21-HK01ba-Vc222-Bf15a-R448a.deb](https://fi.mirror.armbian.de/apt/pool/main/l/linux-6.1.77/linux-image-legacy-bcm2711_24.5.1_arm64__6.1.77-S2101-D440d-Pdbc3-C2812H5c21-HK01ba-Vc222-Bf15a-R448a.deb) and [https://fi.mirror.armbian.de/apt/pool/main/l/linux-6.1.77/linux-image-legacy-bcm2711_24.2.1_arm64__6.1.77-S14c3-D6b4a-Pdbc3-Ca3a2Hfe66-HK01ba-Vc222-B067e-R448a.deb](https://fi.mirror.armbian.de/apt/pool/main/l/linux-6.1.77/linux-image-legacy-bcm2711_24.2.1_arm64__6.1.77-S14c3-D6b4a-Pdbc3-Ca3a2Hfe66-HK01ba-Vc222-B067e-R448a.deb) do not include any dtb files.

KhazAkar, 2024-08-20:
> I was able to build it now.

Can you say how did you managed to do so? My bash-fu is not great, so I'd appreciate any help :)

leggewie, 2025-08-16:
jammy fails in a different way now apparently, but I can confirm the problem for noble target against current HEAD with `./compile.sh RELEASE=noble BOARD=rpi4b BRANCH=legacy NETWORKING_STACK=network-manager KERNEL_CONFIGURE=prebuilt BUILD_DESKTOP=no BUILD_MINIMAL=yes EXTERNAL=yes EXTERNAL_NEW=prebuilt INSTALL_HEADERS=yes DISABLE_IPV6=false COMPRESS_OUTPUTIMAGE=sha,xz KERNEL_GIT=shallow CONSOLE_AUTOLOGIN=no INCLUDE_HOME_DIR=yes`

https://paste.armbian.com/uhoheyolaf