armbian/build #7064: [Bug]: Translate me to carry out the instructions ".../ compile.sh dts-check BOARD=mekotronics-r58x-4x4 BRANCH=vendor ", my dts and dtsi files are automatically deleted, causing me to compile failure

antonyhuang-bit, 2024-08-12:
### What happened?

    I want to compile my newly added DTS file properly and my DTS is not deleted

### How to reproduce?

./compile.sh dts-check BOARD=mekotronics-r58x-4x4 BRANCH=vendor

(base) maizhuo@maizhuo-virtual-machine:~/armbian/build$ ./compile.sh dts-check BOARD=mekotronics-r58x-4x4 BRANCH=vendor
[🌿] Applying cmdline param [ 'BRANCH': '(unset)' --> 'vendor' early ]
[🌿] Applying cmdline param [ 'BOARD': '(unset)' --> 'mekotronics-r58x-4x4' early ]
[🌱] artifact [ kernel :: kernel() ]
[🚸] This script requires root privileges; Docker is unavailable [ trying to use sudo ]
[sudo] password for maizhuo: 
[🌿] Applying cmdline param [ 'PREFER_DOCKER': '(unset)' --> 'no' early ]
[🌱] Skip     cmdline param [ 'ARMBIAN_RELAUNCHED': already set to 'yes' early ]
[🌿] Applying cmdline param [ 'BRANCH': '(unset)' --> 'vendor' early ]
[🌿] Applying cmdline param [ 'BOARD': '(unset)' --> 'mekotronics-r58x-4x4' early ]
[🌿] Applying cmdline param [ 'SET_OWNER_TO_UID': '(unset)' --> '1000' early ]
[🌱] artifact [ kernel :: kernel() ]
[🌱] Starting single build process [ mekotronics-r58x-4x4 ]
[🌱] Checking [ basic host setup ]
[🌱] Build host OS release [ jammy ]
[🌱] Build host architecture [ amd64 ]
[🌱] Sourcing board configuration [ /home/maizhuo/armbian/build/config/boards/mekotronics-r58x-4x4.csc ]
[🌱] shared vendor code [ Mekotronics (RK3588) config ]
[🌿] Enabling extension [ bluetooth-hciattach ]
[✅] change-tracking: after sourcing board file config/boards/mekotronics-r58x-4x4.csc [ BOARDFAMILY='rockchip-rk3588' ]
[✅] change-tracking: after sourcing board file config/boards/mekotronics-r58x-4x4.csc [ BOOTCONFIG='mekotronics_r58x-rk3588-4x4_docker_defconfig mekotronics_r58x-rk3588-4x4_linux_defconfig' ]
[✅] change-tracking: after defaulting LINUXFAMILY to BOARDFAMILY [ LINUXFAMILY='rockchip-rk3588' ]
[🌱] Starting main configuration
[🌱] Using REVISION from [ main VERSION file: '24.8.0-trunk' ]
[🌱] Sourcing family configuration [ /home/maizhuo/armbian/build/config/sources/families/rockchip-rk3588.conf ]
[🌿] Enabling extension [ rkbin-tools ]
[🌱] Determined BOOT_SOC from BOOTCONFIG [ BOOT_SOC: 'rk3588'; BOOTCONFIG: 'mekotronics_r58x-rk3588-4x4_docker_defconfig mekotronics_r58x-rk3588-4x4_linux_defconfig' ]
[✅] change-tracking: after sourcing family config [ KERNELSOURCE='https://github.com/armbian/linux-rockchip.git' ]
[✅] change-tracking: after sourcing family config [ KERNEL_MAJOR_MINOR='6.1' ]
[✅] change-tracking: after sourcing family config [ KERNELBRANCH='branch:rk-6.1-rkr3' ]
[✅] change-tracking: after sourcing family config [ LINUXFAMILY='rk35xx'  # (was: 'rockchip-rk3588') ]
[✅] change-tracking: after sourcing family config [ KERNELPATCHDIR='rk35xx-vendor-6.1' ]
[✅] change-tracking: after sourcing family config [ BOOTSOURCE='https://github.com/radxa/u-boot.git' ]
[✅] change-tracking: after sourcing family config [ BOOTBRANCH='branch:next-dev-v2024.03' ]
[✅] change-tracking: after sourcing family config [ BOOTPATCHDIR='legacy/u-boot-radxa-rk35xx' ]
[✅] change-tracking: after sourcing family config [ BOOTDIR='u-boot-rockchip64' ]
[✅] change-tracking: after sourcing family config [ NETWORKING_STACK='network-manager' ]
[🌱] Sourcing arch configuration [ arm64.conf ]
[🌱] Using NETWORKING_STACK [ NETWORKING_STACK: network-manager ]
[🌱] Adding networking extensions [ net-network-manager, net-chrony ]
[🌿] Enabling extension [ net-network-manager ]
[🌿] Enabling extension [ net-chrony ]
[🌱] Extension manager [ processed 14 Extension Methods calls and 22 Extension Method implementations ]
[🌱] Extension: bluetooth-hciattach: mekotronics-r58x-4x4 [ adding bluetooth packages to image ]
[🌱] mekotronics-r58x-4x4 [ Configuring mekotronics-r58x-4x4 vendor u-boot ]
[🌱] rockchip64_common: defaulting SERIALCON [ Setting SERIALCON to ttyFIQ0 for BRANCH='vendor' ]
[🌱] Extension: bluetooth-hciattach: mekotronics-r58x-4x4 [ initializing config ]
[🌱] Extension: net-chrony: Adding extra package to image [ chrony ]
[🌱] Extension: net-network-manager: Adding extra packages to image [ network-manager network-manager-openvpn netplan.io ]
[✅] change-tracking: after late_family_config hooks [ LINUXCONFIG='linux-rk35xx-vendor' ]
[✅] change-tracking: after late_family_config hooks [ KERNEL_PATCH_ARCHIVE_BASE='rk35xx' ]
[✅] change-tracking: before handling KERNEL_MAJOR_MINOR in config_post_main [ BOOTSOURCEDIR='u-boot-worktree/u-boot-rockchip64/next-dev-v2024.03' ]
[✅] change-tracking: before calling extension_finish_config [ LINUXSOURCEDIR='linux-kernel-worktree/6.1__rk35xx__arm64' ]
[🌱] Minimal configuration prepared for build [ prep_conf_main_minimal_ni ]
[✨] Repeat Build Options (early) [ ./compile.sh dts-check BOARD=mekotronics-r58x-4x4 BRANCH=vendor ]
[🌱] Checked directory OK for mount options [ /home/maizhuo/armbian/build/.tmp ('main temporary dir') ]
[🌱] Preparing [ host ]
[🌱] Syncing clock [ host ]
[🔨]   12 Aug 10:12:14 ntpdate[4534]: adjust time server 202.112.29.82 offset -0.046937 sec
[🌱] Ignoring toolchains [ SKIP_EXTERNAL_TOOLCHAINS: yes ]
[🌱] Setting kernel git cache TTL to [ 120 ]
[🌱] Producing new & caching [ GIT_INFO_KERNEL ]
[... 92 lines cut ...]
               cli_artifact_run() --> lib/functions/cli/cli-artifact.sh:67
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
[🌿] ANSI log file built; inspect it by running: [ less -RS output/logs/log-dts-check-9bed64ed-e839-48fb-b59b-a53fe236bef5.log.ans ]
[🌿] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-dts-check-9bed64ed-e839-48fb-b59b-a53fe236bef5.log.ans https://paste.armbian.com/log ]
(base) maizhuo@maizhuo-virtual-machine:~/armbian/build$ 

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-08-12:
Is this device already in the system?

antonyhuang-bit, 2024-08-12:
> Is this device already in the system?该设备是否已在系统中？

Thank you for your reply ,  This device has not been added to the system yet, I am trying to add it

igorpecovnik, 2024-08-12:
>  I am trying to add it

Understand.

We don't have the best manual (they are in fact terrible) but try to study it, code, examples, try command `kernel-patch`, add code to the sources with a patch or put DT directly ...

SuperKali, 2024-09-06:
> I am trying to add it

The fastest method is to fork the kernel on your GitHub profile, add the dts and related changes, clone this repository  https://github.com/armbian/build and edit the `config/sources/families/rockchip-rk3588.conf` file changing the kernel pointers to your repository.

leggewie, 2025-04-27:
@antonyhuang-bit do you still need help with this?

leggewie, 2025-06-22:
closing as stale for now

feel free to reopen at any time as necessary

closed 2025-06-22 by leggewie