armbian/build #7087: [Bug]: Failed to apply patch /armbian/patch/kernel/archive/rockchip64-6.6/drv-spi-spidev-remove-warnings.patch

clarkmcc, 2024-08-14:
### What happened?

My build is getting this error all of the sudden. It looks like it pulled down a new version of the kernel try (can I pin that to a specific version in my compile.sh script?). 

### How to reproduce?

```
compile.sh build BOARD=nanopi-r2s BRANCH=current BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm ROOTFS_TYPE=btrfs SHARE_LOG=yes
```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/ifitonaxox

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-08-14:
> It looks like it pulled down a new version of the kernel

This is default behavior. Sources are frozen at [release branches](https://github.com/armbian/build/commit/bca58a204ec6046ba36d8517c56585513dc75880) or if you set this in config file.

clarkmcc, 2024-08-14:
@igorpecovnik okay, but that was just an ancillary part of my question. The main question was this error, and I've added a link to the full log file. Is this not a bug?
```
Failed to apply patch /armbian/patch/kernel/archive/rockchip64-6.6/drv-spi-spidev-remove-warnings.patch
```

EvilOlaf, 2024-08-14:
Try from 24.08 branch which should have the mentioned frozen versions.
Or try the fix from @alexl83 if you want most recent available.

clarkmcc, 2024-08-14:
@EvilOlaf thanks!

closed 2024-08-14 by EvilOlaf

igorpecovnik, 2024-08-14:
> Is this not a bug?

Yes, but this bug is as yours as its ours / mine. Its not a framework bug.