armbian/build #7158: [Feature Request]: Add support for Banana Pi BPI-M4

schrieveslaach, 2024-09-01:
### Which feature would you like to have?

I'm trying to mow my lawn and I bought the [DIY lawn mower Alfred](https://alfred.marotronics.de/). Later I discovered that it is running on a [BananaPI M4](https://docs.banana-pi.org/en/BPI-M4/BananaPi_BPI-M4) which isn't yet supported by a nice ARM Debian based distro. I want to improve the software update process (see https://github.com/Ardumower/Sunray/issues/163) and thus I would like to have Armbian support.

| Feature | Spec |
| ---- | ---- |
| CPU  | Realtek RTD1395 ARM Cortex-A53 Quad-Core 64 Bit |
| GPU  | Mali 470 MP4 GPU OpenGL ES 1.1/2.0 |
| Memory  | 1 GB DDR4 (option 2 GB) |
| Storage  | MicroSD slot with support for up to 256GB expansion and 8G eMMC flash with support for up to 64GB |
| Network  | 10/100 Mbit/s Ethernet + Wi-Fi 802.11 b/g/n/AC + Bluetooth 4.2 |
| Video Output(s)  | HDMI port and multi-channel audio output support 1080P H.264 H.265 2K/4K |
| Audio Output(s)  | 3.5mm jack and HDMI |
| PCIE  | M.2 Key E slot PCIE 2.0 and USB 2.0 |
| USB ports  | USB 2.0 PORT (x4), USB 2.0 TYPE C(x1) |
| GPIO  | 40 Pin Header : GPIO (x28) and Power (+5V, +3.3V and GND). GPIO pins can be used for UART, I2C, SPI or PWM |
| Switches  | Reset, Power and U-boot |
| LED  | Power Status and Activity status |
| Power Source  | 5 volt @2A via Micro USB (TYPE C) or PoE support |

I'm willing to support here in some form (not sure if I [become a maintainer](https://docs.armbian.com/Board_Maintainers_Procedures_and_Guidelines/) or if I want to cover the expenses).

Maybe, someone can help me out to get started (Already trying to create a `csc` file).

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

EvilOlaf, 2024-09-01:
Hi,

Armbian will not consider this request and maintenance without funding.
However anyone from the community is free to donate their time adding support for new hardware.

schrieveslaach, 2024-09-02:
@EvilOlaf, I was looking for a source of how much one would need to spend to cover the expenses but I was unable to find any (other then [why your project needs money](https://docs.armbian.com/User-Guide_FAQ/#why-is-armbian-constantly-asking-for-money-free-software-should-be-free) for good reason, IMHO). Do you have any pointers? It doesn't have to be accurate.

igorpecovnik, 2024-09-02:
> Is that accurate?

This is a raw estimation. It can be half less and several times more, a million can also easily be spent. Depends what is the status now and what is the goal. Perhaps we can't even deliver so we will say "sorry, this too big for us" "this far we can go" ... once we know more. Its a software development at currently unknown field. Every SoC is different beast and Armbian is "just" a tool that outputs Linux OS. We are talking about SoC that has at this point unknown support. Estimating this is a small project and not so small expense. If you want to proceed, good first step would be to start some form of deal / partnership to cover meetings and research to gathered enough that we know what we are looking at. If we can't do it, our research will be valuable input for someone that can do it. Research is expense of both right from start or there is no go. If you see a potential and can finance discussions and later development, perhaps this [should be a good way to start](https://calendly.com/armbian/consultation).

schrieveslaach, 2024-09-02:
@igorpecovnik, thanks for info. Would be nice to have this info in the FAQ or the link behind the checkbox “[I am willing to cover expenses](https://www.armbian.com/contact/)” could provide this information.

I think your explanation and reasoning is valid. However, that is way beyond my financial capabilities. :disappointed: 

Maybe, I try to be helpful to the community and gather some information and maybe, try to support the board.

schrieveslaach, 2024-09-02:
@afaerber, I discovered that you brought support for BPI M4 into Linux mainline (see https://github.com/torvalds/linux/commit/769c00a2f10b4f43fe764077a48d9a594010686b). Are you aware of any Linux distribution that supports the Bananapi M4 out of the box?

And sorry, for being dragged into, at least from your perspective, a random GitHub discussion.

igorpecovnik, 2024-09-03:
> Would be nice to have this info in the FAQ

This is still half solution at best. A lot of people don't read FAQ and there other half ignores this anyway ;) Not sure if its worth. We usually just close the ticket without explanation. If you don't know what you are asking ... 

A lot would be nice to do communication wise, but we don't have capacity. This is extreme expense - and nothing for this on the other side - is hard to patch. [Improving documentation](https://github.com/armbian/documentation/issues/467) is volunteer based. And we provide reward to you.

> that is way beyond my financial capabilities

I totally understand you. Same is for us. Cost is not small and once hardware is brought to Armbian, anyone can use it without investing anything. You don't care about that, but it is self destructive for us.

It is a lot worse for us as you (and competitors that will invest nothing providing the same) expects that hardware remains maintained. Adding is just a firsts step.

> I discovered that you brought support for BPI M4 into Linux mainline

This means little. Most of hardware support in mainline Linux needs (heavy) patching in order to work. This only means some base work has been done, but it can be far away from full functionality.

Try this way:
https://docs.armbian.com/Developer-Guide_Adding-Board-Family/

schrieveslaach, 2024-10-12:
Update: I tried to figure out what I have to do to bring Armbian support to this board but I think that is beyond my capabilities and the time I can spent for trying it. I think I have a good gut feeling why you demand 20k € per as a donation.

For me the only feasible option at the moment is to switch to a Raspberry PI, like others do: https://forum.ardumower.de/threads/sunray-installation-auf-raspberry.25408/post-64382

@greymfm, do you see an option to collect the required amount of money via a call to action in the forum?

igorpecovnik, 2024-10-12:
> For me the only feasible option at the moment is to switch to a Raspberry PI, like others do

There are many options, just not the hardware you want.

We **recommend** to use one of these:
https://www.armbian.com/download/?device_support=Platinum%20support

Still good anything from this list:
https://www.armbian.com/download/?device_support=Standard%20support

Most of those will also work, but no warranty as we don't monitor them, but a lot cheaper to sort them out:
https://www.armbian.com/download/?device_support=Community%20maintained

> you demand 20k € per as a donation.

Try not to mix donations here. Donations are free willing and unconditional response to, primarily, volunteering work that was already done.

This 20k is blind estimation of costs you will have with your problem, not a business offer. For that you need to [start here](https://www.armbian.com/contact/) and become a Gold partner. However there is no warranty that we will proceed, but there are companies, specialised in bringing up Linux to random hardware, to help out.

leggewie, 2025-08-24:
closing this as wontfix for now