armbian/build #7178: [Bug]: Main Temporary Dir Mount Error

chemrgineer, 2024-09-04:
### What happened?

I am using Linux Mint 21.3 and for fresh start I cloned repo with `git clone --depth=1 --branch=main https://github.com/armbian/build`. Than I followed compile.sh instructions. I selected all basic options for Radxa Zero board. I'm attaching [terminal output](https://github.com/user-attachments/files/16879157/build_error_log.txt). 
I get `Directory /home/chemrgineer/radxazero/build/.tmp (main temporary dir) is mounted with the 'noexec' and/or 'nodev' options; this will cause rootfs build failures. Please correct this before trying again.` I tried with sudo previliges nothing changed. 
Note: SHARED_LOG=yes option did not worked.

### How to reproduce?

Get latest repo with:
`git clone --depth=1 --branch=main https://github.com/armbian/build`
get into repo directory:
`cd ./build`
For fast error reproduction just give all options in once:
`./compile.sh build BOARD=radxa-zero BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no RELEASE=noble`

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-09-05:
I guess there is an incompatibility with ecryptfs. Try a different, non-encrypted, directory.

SuperKali, 2024-09-06:
This is the official steps to compile Armbian, please follow this steps: https://docs.armbian.com/Developer-Guide_Build-Preparation/  or use docker image for compile it

chemrgineer, 2024-09-06:
Yes, I use encrypted home directory in Linux Mint. I tried same thing in WLS2 it works. I could not figure out from error log. Thanks for help.

closed 2024-09-06 by chemrgineer