armbian/build #7185: [Bug]: WiFi connection numbering is wrong

Kreyren, 2024-09-07:
### What happened?

[image]

hidden networks seems to mess up the count so that if i want to choose wifi indexed as `5` i have to write `6`

### How to reproduce?

Try to perform the initial installation on a network with hidden SSID

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-09-26:
Is this good enough?
https://github.com/armbian/build/commit/31e88922ad2b7a387c30ca9fbda618306dbd638b

leggewie, 2025-06-22:
> ### What happened?
> [image]

what command is that?

leggewie, 2025-06-23:
no, the entire screenshot.  what command generated that output?