armbian/build #7186: [Bug]: GPG error signatures were invalid: EXPKEYSIG 23F3D4EA75716059

gufeisheng001, 2024-09-08:
### What happened?

while compiling get an
[image]
GPG error: https://cli.github.com/packages stable InRelease: The following signatures were invalid: EXPKEYSIG 23F3D4EA75716059 GitHub CLI <opensource+cli@github.com>

### How to reproduce?

./compile.sh build BOARD=orangepi5-plus BRANCH=current BUILD_DESKTOP=yes BUILD_MINIMAL=no DESKTOP_APPGROUPS_SELECTED='browsers chat desktop_tools editors email internet multimedia office programming remote_desktop' DESKTOP_ENVIRONMENT=gnome DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base KERNEL_CONFIGURE=yes RELEASE=bookworm
[log-build-84f3f0fb-a224-4da9-a1a6-c709fe82d013.log](https://github.com/user-attachments/files/16920946/log-build-84f3f0fb-a224-4da9-a1a6-c709fe82d013.log)

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-09-08:
We need to wait for GH to replace their expired key

https://github.com/cli/cli/issues/9569

closed 2024-09-08 by EvilOlaf

williammartin, 2024-09-08:
Hi from the GitHub CLI team and very sorry for the inconvenience here. 

I see this is already closed out but there seems to be some confusion:

> We need to wait for GH to replace their expired key https://github.com/cli/cli/issues/9569

The key was already updated on Thursday 5th September. If you are still running into issues let me know and I'll try to provide support here.

gufeisheng001, 2024-09-08:
> Hi from the GitHub CLI team and very sorry for the inconvenience here.
> 
> I see this is already closed out but there seems to be some confusion:
> 
> > We need to wait for GH to replace their expired key [cli/cli#9569](https://github.com/cli/cli/issues/9569)
> 
> The key was already updated on Thursday 5th September. If you are still running into issues let me know and I'll try to provide support here.

Hi, i did those step, after that run ./compile.sh (even got an fresh install Ubuntu22.04.1LTS to test it out) but still got the same error.
what i did is :
1) wget -qO- https://cli.github.com/packages/githubcli-archive-keyring.gpg | sudo tee /etc/apt/keyrings/githubcli-archive-keyring.gpg > /dev/null
2) sudo chmod go+r /etc/apt/keyrings/githubcli-archive-keyring.gpg
3) echo "deb [arch=$(dpkg --print-architecture) signed-by=/etc/apt/keyrings/githubcli-archive-keyring.gpg] https://cli.github.com/packages stable main" | sudo tee /etc/apt/sources.list.d/github-cli.list > /dev/null
4) sudo apt update
gh is already the newest version (2.55.0).
So did i miss some step, or other things?

williammartin, 2024-09-08:
When you run `sudo apt update` do you see the same error?

>  GPG error signatures were invalid: EXPKEYSIG 23F3D4EA75716059

If you're seeing `gh is already the newest version (2.55.0).` it seems like you have the correct keyring. If you're seeing an different error with the project, I'm not sure what that would be.

gufeisheng001, 2024-09-08:
>Nope, apt update did not see errors, only saw it while compiling Armbian build
[image]

williammartin, 2024-09-08:
Ah, I'm not sure what that is doing but it seems likely that [this file](https://github.com/armbian/build/blob/56f5d6692c08a62dcf698a57eaf06963e5f2b67a/config/desktop/_all_distributions/appgroups/programming/sources/apt/githubcli.gpg) needs to be updated with https://cli.github.com/packages/githubcli-archive-keyring.gpg

I created a PR here https://github.com/armbian/build/pull/7192 with the change I think is required.

EvilOlaf, 2024-09-09:
Apologies. Did a quick search across the framework but could not find it. So it was overall an issue on our end.

gufeisheng001, 2024-09-09:
I got mistake the default gpg file location on debian/ubuntu, that still at /usr/share/keyrings/ NOT /etc/apt/keyrins/. Based on the https://github.com/cli/cli/issues/9569#issuecomment-2333981674
no errors present any more after done those steps
1) wget -qO- https://cli.github.com/packages/githubcli-archive-keyring.gpg | sudo tee /usr/share/keyrings/githubcli-archive-keyring.gpg > /dev/null
2) sudo chmod go+r /usr/share/keyrings/githubcli-archive-keyring.gpg
3) echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/githubcli-archive-keyring.gpg] https://cli.github.com/packages stable main" | sudo tee /etc/apt/sources.list.d/github-cli.list > /dev/null

THANKS ALL!

williammartin, 2024-09-10:
Ah, sorry about that @gufeisheng001, after that comment we updated the original issue description [with a script to handle that difference](https://github.com/cli/cli/issues/9569#what-do-you-need-to-do-about-apt) but it was probably too late for you. Glad you have it fixed.