armbian/build #7207: [Bug]: mali_csffw.bin can't be loaded

DavidHenryThoreau, 2024-09-11:
### What happened?

The driver `mali_csffw.bin` can't be loaded

### How to reproduce?

```
cp /lib/firmware/arm/mali/arch10.8/mali_csffw.bin /lib/firmware

 docker exec -it jellyfin /usr/lib/jellyfin-ffmpeg/ffmpeg -v debug -init_hw_device rkmpp=rk -init_hw_device opencl=ocl@rk
ffmpeg version 6.0.1-Jellyfin Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-static --disable-libxcb --disable-sdl2 --disable-xlib --enable-lto --enable-gpl --enable-version3 --enable-shared --enable-gmp --enable-gnutls --enable-chromaprint --enable-opencl --enable-libdrm --enable-libxml2 --enable-libass --enable-libfreetype --enable-libfribidi --enable-libfontconfig --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libopenmpt --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libfdk-aac --arch=arm64 --cross-prefix=/usr/bin/aarch64-linux-gnu- --toolchain=hardened --enable-cross-compile --enable-rkmpp --enable-rkrga
  libavutil      58.  2.100 / 58.  2.100
  libavcodec     60.  3.100 / 60.  3.100
  libavformat    60.  3.100 / 60.  3.100
  libavdevice    60.  1.100 / 60.  1.100
  libavfilter     9.  3.100 /  9.  3.100
  libswscale      7.  1.100 /  7.  1.100
  libswresample   4. 10.100 /  4. 10.100
  libpostproc    57.  1.100 / 57.  1.100
Splitting the commandline.
Reading option '-v' ... matched as option 'v' (set logging level) with argument 'debug'.
Reading option '-init_hw_device' ... matched as option 'init_hw_device' (initialise hardware device) with argument 'rkmpp=rk'.
Reading option '-init_hw_device' ... matched as option 'init_hw_device' (initialise hardware device) with argument 'opencl=ocl@rk'.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option v (set logging level) with argument debug.
Applying option init_hw_device (initialise hardware device) with argument rkmpp=rk.
Applying option init_hw_device (initialise hardware device) with argument opencl=ocl@rk.
arm_release_ver: g13p0-01eac0, rk_so_ver: 10
[AVHWDeviceContext @ 0xaaaac71b4f00] 1 OpenCL platforms found.
[AVHWDeviceContext @ 0xaaaac71b4f00] 1 OpenCL devices found on platform "ARM Platform".
[AVHWDeviceContext @ 0xaaaac71b4f00] 0.0: ARM Platform / Mali-G610 r0p0
[AVHWDeviceContext @ 0xaaaac71b4f00] cl_arm_import_memory found as platform extension.
[AVHWDeviceContext @ 0xaaaac71b4f00] cl_khr_image2d_from_buffer found as platform extension.
[AVHWDeviceContext @ 0xaaaac71b4f00] DRM to OpenCL mapping on ARM function found (clImportMemoryARM).
Successfully parsed a group of options.
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'

dmesg|grep Mali
[  367.986293] mali fb000000.gpu: Loading Mali firmware 0x1010000
[  367.988606] mali fb000000.gpu: Mali firmware git_sha: 814b47b551159067b67a37c4e9adda458ad9d852
```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amazingfate, 2024-09-11:
How do you know it is not loaded?
FFMpeg has already outputed opencl info.

closed 2024-09-11 by amazingfate

DavidHenryThoreau, 2024-09-11:
Because If I don't copy `cp /lib/firmware/arm/mali/arch10.8/mali_csffw.bin /lib/firmware` firmware can't be loaded
Test to reproduce

DavidHenryThoreau, 2024-09-11:
Because If I don't copy `cp /lib/firmware/arm/mali/arch10.8/mali_csffw.bin /lib/firmware` firmware can't be loaded
I've edited first post

amazingfate, 2024-09-11:
This is normal. Armbian doesn't provide firmware for arm mali driver. User has to deal with it by themselves.

DavidHenryThoreau, 2024-09-11:
The firmware is located in `/lib/firmware/arm/mali/arch10.8/mali_csffw.bin` after fresh installed

amazingfate, 2024-09-11:
> The firmware is located in `/lib/firmware/arm/mali/arch10.8/mali_csffw.bin` after fresh installed

It is for mainline panthor driver. For arm mali driver it is more complicated.

DavidHenryThoreau, 2024-09-11:
Yes, but it seems to be loaded.
Do you plan to add arm mali driver for next release ?

amazingfate, 2024-09-11:
No, different kernel driver DDK need different versions of firmware. One file make work for some kernel, but not all. It's a hell for maintaining.

DavidHenryThoreau, 2024-09-11:
Ok
And add the firmware in debian package in external repository ?

amazingfate, 2024-09-11:
I have packaged firmware package for bookworm: https://software.opensuse.org//download.html?project=home%3Aamazingfate%3Apanfork-mesa&package=mali-g610-firmware

DavidHenryThoreau, 2024-09-11:
Nice thanks a lot !
I'm gonna try it !