armbian/build #7214: [Bug]: Additional Overlays in patch dir are not built

defencedog, 2024-09-11:
### What happened?

Additional `dtbo` must be generated & packed into resulting `deb`
No `dtbo` were generated

### How to reproduce?

My [0000.patching_config.yaml](https://github.com/rpardini/armbian-build/blob/extensions/patch/kernel/archive/rk322x-6.1/0000.patching_config.yaml)
```
config:

  # Just some info stuff; not used by the patching scripts
  name: rk35xx-6.1
  kind: kernel
  type: vendor # or: vendor
  branch: rk-6.1-rkr3
  last-known-good-tag: v6.1.75

  # .dts files in these directories will be copied as-is to the build tree; later ones overwrite earlier ones.
  # This is meant to provide a way to "add a board DTS" without having to null-patch them in.
  dts-directories:
    - { source: "dt", target: "arch/arm64/boot/dts/rockchip" }

  overlay-directories:
    - { source: "overlay", target: "arch/arm/boot/dts/rockchip/overlay" }

  # the Makefile in each of these directories will be magically patched to include the dts files copied
  #  or patched-in; overlay subdir will be included "-y" if it exists.
  # No more Makefile patching needed, yay!
  # "incremental: true" changes the logic of the Makefile re-writing to only add the
  #                     dts-directories's *.dts files to existing Makefile instead of
  #                     full rewrite from *.dts in the dt dir at the end of patching.
  auto-patch-dt-makefile:
    - { incremental: yes, directory: "arch/arm64/boot/dts/rockchip", config-var: "CONFIG_ARCH_ROCKCHIP" }

  # configuration for when applying patches to git / auto-rewriting patches (development cycle helpers)
  patches-to-git:
    do-not-commit-files:
      - "MAINTAINERS" # constant churn, drop them. sorry.
      - "Documentation/devicetree/bindings/arm/rockchip.yaml" # constant churn, conflicts on every bump, drop it. sorry.
    do-not-commit-regexes: # Python-style regexes
      - "^arch/([a-zA-Z0-9]+)/boot/dts/([a-zA-Z0-9]+)/Makefile$" # ignore DT Makefile patches, we've an auto-patcher now
```

then file locations
```
:~/build/patch/kernel/rk35xx-vendor-6.1/overlay$ ls
Makefile                  rk356x-edp.dts  rk356x-i2c2-m1.dts  rk356x-i2c4-m0.dts    rk356x-ov5647-c2.dts  rk356x-pwm15-m1.dts  rk356x-spi3-m0-cs0-spidev.dts  rk356x-uart7-m2.dts
README.rockchip-overlays  rk356x-gpu.dts  rk356x-i2c3-m0.dts  rk356x-ov5647-c1.dts  rk356x-pwm11-m1.dts   rk356x-pwm7.dts      rk356x-uart3-m0.dts            rk356x-uart9-m2.dts
```

contents of Makefile in overlay folder
```
# SPDX-License-Identifier: GPL-2.0
dtbo-$(CONFIG_ARCH_ROCKCHIP) += \
        rk356x-uart9-m2.dtbo\
        rk356x-uart7-m2.dtbo\
        rk356x-uart3-m0.dtbo\
        rk356x-i2c2-m1.dtbo\
        rk356x-i2c3-m0.dtbo\
        rk356x-i2c4-m0.dtbo\
        rk356x-pwm11-m1.dtbo\
        rk356x-pwm15-m1.dtbo\
        rk356x-spi3-m0-cs0-spidev.dtbo\
        rk356x-gpu.dtbo

dtbotxt-$(CONFIG_ARCH_ROCKCHIP) += \
        README.rockchip-overlays

targets += $(dtbo-y) $(dtbotxt-y)

always         := $(dtbo-y) $(dtbotxt-y)
clean-files    := *.dtbo

```

Per the progress overlay files are added to correct location  but `dtbo` not generated. This is verified by `dpkg --contents xx.deb`

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [X] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/zupigobeyi 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amazingfate, 2024-09-12:
The method you use is for mainline kernel, which doesn't have overlay dir, so just copying Makefile and all dts files would work. Vendor kernel has overlay dir, and adding overlay dts to Makefile is not implemented.

defencedog, 2024-09-12:
@amazingfate so please can you provide an alternative? Any changes I have to make `0000.patching_config.yaml` or manual patching elsewhere...

amazingfate, 2024-09-13:
You can create patch based on kernel source and add the patch to `patch/kernel/rk35xx-vendor-6.1`

leggewie, 2025-04-27:
@defencedog did amazingfate's suggestion solve your issue?

closed 2025-05-10 by defencedog