armbian/build #7221: [Bug]: orangepipcplus cannot set the antenna power.

Char11e, 2024-09-13:
### What happened?

orangepipcplus cannot set the antenna power.

### How to reproduce?

`pi@orangepipcplus:~$ uname -a
Linux orangepipcplus 6.6.44-current-sunxi #1 SMP Sat Aug  3 06:54:42 UTC 2024 armv7l armv7l armv7l GNU/Linux
pi@orangepipcplus:~$ ^C
pi@orangepipcplus:~$ sudo iw dev wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr 12:81:64:17:11:3d
        ssid R15-3437
        type managed
        wiphy 0
        txpower 12.00 dBm
pi@orangepipcplus:~$ sudo iw dev wlan0 set txpower fixed 1500
pi@orangepipcplus:~$ sudo iw dev wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr 12:81:64:17:11:3d
        ssid R15-3437
        type managed
        wiphy 0
        txpower 12.00 dBm
`

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Char11e, 2024-09-14:
'pi@orangepipcplus:~$ uname -a'
'Linux orangepipcplus 6.6.44-current-sunxi #1 SMP Sat Aug  3 06:54:42 UTC 2024 armv7l armv7l armv7l GNU/Linux'
'pi@orangepipcplus:~$ cat /etc/os-release'
'PRETTY_NAME="Armbian 24.8.0-trunk.581 jammy"'
'NAME="Ubuntu"'
'VERSION_ID="22.04"'
'VERSION="22.04.4 LTS (Jammy Jellyfish)"'
'VERSION_CODENAME=jammy'
'ID=ubuntu'
'ID_LIKE=debian'
'HOME_URL="https://www.armbian.com"'
'SUPPORT_URL="https://forum.armbian.com"'
'BUG_REPORT_URL="https://www.armbian.com/bugs"'
'PRIVACY_POLICY_URL="https://www.armbian.com"'
'UBUNTU_CODENAME=jammy'
'ARMBIAN_PRETTY_NAME="Armbian 24.8.0-trunk.581 jammy"'

Char11e, 2024-09-14:
`pi@orangepipcplus:~$ sudo lshw -C network`
`  *-network:0`
`       description: Ethernet interface`
`       physical id: 12`
`       logical name: eth0`
`       serial: 02:81:64:17:11:3d`
`       size: 100Mbit/s`
`       capacity: 100Mbit/s`
`       capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd autonegotiation`
`       configuration: autonegotiation=on broadcast=yes driver=st_mac100 driverversion=6.6.44-current-sunxi duplex=full ip=192.168.1.100 link=yes multicast=yes port=twisted pair speed=100Mbit/s`
`  *-network:1`
`       description: Wireless interface`
`       physical id: 13`
`       bus info: mmc@1:0001:1`
`       logical name: wlan0`
`       serial: 12:81:64:17:11:3d`
`      capabilities: ethernet physical wireless`
`       configuration: broadcast=yes driver=rtl8189fs driverversion=6.6.44-current-sunxi ip=192.168.0.214 multicast=yes wireless=IEEE 802.11bgn`
`  *-network:2`
`       description: Wireless interface`
`       physical id: 14`
`       bus info: mmc@1:0001:1`
`       logical name: wlan1`
`       serial: 12:81:64:17:13:3d`
`       capabilities: ethernet physical wireless`
`       configuration: broadcast=yes driver=rtl8189fs driverversion=6.6.44-current-sunxi multicast=yes wireless=unassociated`

Char11e, 2024-09-14:
I'm not sure if this is related to a limitation of the rtl8189fs firmware.

EvilOlaf, 2024-09-14:
Not a framework bug
https://armbian.com/bugs

closed 2024-09-14 by EvilOlaf