armbian/build #7234: [Bug]: syntax error in /etc/update-motd.d/30-armbian-sysinfo

CT1IQI, 2024-09-14:
### What happened?

30-armbian-sysinfo produces system info upon login, but the function display() can receiver as parameter $2 a floating point number from the traffic reporting function. Bash only knows integer arithmetic. In line
if [[ -n "$2" && "$2" -gt "0" && (( "${2%.*}" -ge "$4" )) ]]; then
The '$2 -gt' throws an error. The second -ge has been corrected for not having the dot decimal part of the string.
This should also be done for the first: 
if [[ -n "$2" && "${2%.*}" -gt "0" && (( "${2%.*}" -ge "$4" )) ]]; then

second problem:
hard coded in line 19 is the interface name eth0. But Armbian changed this to end0.

### How to reproduce?

run the script

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-09-14:
> hard coded in line 19 is the interface name eth0. But Armbian changed this to end0.

This is read from /etc/default/armbian-motd where we have:

PRIMARY_INTERFACE="$(ip route | grep '^default' | sed "s/.*dev //" | cut -d" " -f1 | head -1)"

Fixing is coming in a minute.

SuperKali, 2024-09-17:
@CT1IQI should be fixed with #7247 can you retry with compile a newer version?

SuperKali, 2024-09-25:
The issue should be resolved by Igor, i will close this issue

closed 2024-09-25 by SuperKali