armbian/build #7269: [Bug]: compile error using vendor rk3588 6.1 kernel on

CT1IQI, 2024-09-17:
### What happened?

compiling image for rk3588s system, compilations halts with error:
/usr/bin/aarch64-linux-gnu-ld: vmlinux.a: member fs/lockd/procfs.o in archive is not an object
make[2]: *** [scripts/Makefile.vmlinux_o:61: vmlinux.o] Error 1
make[1]: *** [Makefile:1236: vmlinux_o] Error 2
make: *** [Makefile:369: __build_one_by_one] Error 2
Error 2 occurred in main shell [ at /home/armbian/build/lib/functions/logging/runners.sh:211

 

### How to reproduce?

Reverted from compiling with edge kernel for rk3588s radxa cm5 based board from edge to vendor because of lacking dsi lcd drivers in edge. compiling for edge as such worked, for vendor kernel above error emerges.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

SuperKali, 2024-09-17:
Can attach log of compile?

EvilOlaf, 2024-09-17:
Agree. No logs, no issue.

SuperKali, 2024-09-17:
I can't reproduce your problem on my work environment:

https://paste.armbian.com/ijoxacihot

closed 2024-09-17 by SuperKali

CT1IQI, 2024-09-17:
@SuperKali,
thanks, went back to edge compilation and in other spot same kind of error.
Am currently loading a fresh install of build as probably something in the build system was damaged.
Will report back, with log, if this resurfaces.