armbian/build #7270: [Feature Request]: Enable CONFIG_DEBUG_INFO & CONFIG_DEBUG_INFO_BTF

Integral-Tech, 2024-09-17:
### Which feature would you like to have?

It would be better if armbian kernels can enable ```CONFIG_DEBUG_INFO``` & ```CONFIG_DEBUG_INFO_BTF```, as ```CONFIG_DEBUG_INFO=y``` & ```CONFIG_DEBUG_INFO_BTF=y``` are indispensable for [dae](https://github.com/daeuniverse/dae) (an eBPF-based Linux high-performance transparent proxy solution).

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

rpardini, 2024-09-18:
This is in the works for a few months now. I've a [working-ish version on my branch](https://github.com/armsurvivors/armbian-build/commit/9e1c89f575f4ad3adeaabc2c1ad0a28cd9f51fff) that needs some cleanup; @Integral-Tech you might be able to pick it into your own userpatches/config.

Enabling full BTF debug info (such as it's CO-RE compatible) puts about 2x more load on the kernel build servers and increases the build time by around 50%, not to mention a small increase in the size of the kernel packages and a _huge_ (2x-3x) increase in disk consumption for building kernels.

As I've been unable to find workarounds for those increases, and we've been unable to increase build server capacity, I hesitate to send the PR.

@igorpecovnik what do you think?

igorpecovnik, 2024-09-18:
That means cca. + additional 256 CPU cores to keep the same compile time and storage upgrades on some machines that are on the edge (cca. half of them). If bulk of those expenses are not on us, I don't mind.

closed 2024-10-29 by Integral-Tech