armbian/build #7275: [Feature Request]: When creating output in 'packages_hashed', and 'debs', remove previous of same specification

CT1IQI, 2024-09-17:
### Which feature would you like to have?

A mechanism that does clean up between builds of in principle the same image. Am working inside a Virtual Box with ample, but limited, room. When creating a customized image it is normal to have to go through some iterations. Every round the build system adds quite substantial files (~300 MB in my case) in output/packages-hashed, and output/debs, without removing the ones left behind by previous sessions. After a while the available disk memory gets filled up this way with large and useless files. 

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

rpardini, 2024-09-18:
That is by design; as versions can change more frequently than the content of the packages. 
You can clean the `output` folder as you wish (and pay for recompiling/redownloading them the next time) if you don't have the disk space.

CT1IQI, 2024-09-18:
Tnx. The issue is that between compiles there are no changes in the Ubuntu packages (Armbian defined plus some 145 pcs. user defined) that the image needs to contain. Only changes in the userpatches section to rearrange rootfs things. So the idea would be, if the needed packages are already available in their right version, to re-use them, construct the new 'big' file, but then to delete the big one from which the re-used packages were imported. That way only the latest and used versions of packages remain on board, not the complete history.