armbian/build #7282: Deprecate `WIREGUARD`?

EvilOlaf, 2024-09-19:
### Which feature would you like to have?

https://github.com/search?q=repo%3Aarmbian%2Fbuild%20wireguard-tools&type=code
It seems like wireguard-tools is already in the list of common packages and `WIREGUARD=yes` basically installs this package. 
So besides using it on a minimal image there is no real use for it. So maybe deprecate `WIREGUARD` entirely now?

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

igorpecovnik, 2024-09-19:
> So maybe deprecate WIREGUARD entirely now?

Probably best. And deal with this here https://github.com/armbian/configng

EvilOlaf, 2024-09-20:
There are a handful of board config files with `WIREGUARD=no` but I assume this line does not do anything since the `wireguard-tools` package is part of the common list.
Besides those removal is necessary here https://github.com/armbian/build/blob/5e8d48cb7a1202336e2e3cf3b89aef2a0f1b9c09/lib/functions/configuration/main-config.sh#L266
and here https://github.com/armbian/build/blob/5e8d48cb7a1202336e2e3cf3b89aef2a0f1b9c09/lib/functions/rootfs/distro-agnostic.sh#L357

And that should be pretty much it.

closed 2024-11-09 by igorpecovnik