armbian/build #7295: [Feature Request]: Luckfox Core3566 ugly support or to not support at all

buldo, 2024-09-24:
### Which feature would you like to have?

Hello, there is RK3566 compute module Luckfox Core3566. As I understand it co-branded in some way with Waveshare.  
Manufacturer provides very bad sdk based on old kernel.  

What I did:
+ take armbian config files and u-boot dts from radxa zero3 and modified it
+ added dts from Luckfox SDK kernel

This allows me to get working image for Luckfox Core3566( my board has no wifi module and emmc).

But patch soooo ugly so I don't know to do or not PR

[armbian-core3566.patch](https://github.com/user-attachments/files/17121918/armbian-core3566.patch)

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

EvilOlaf, 2024-09-25:
The dts files should go into armbian/linux-rockchip like this: https://github.com/armbian/linux-rockchip/pull/245 and the rest into the build repo here.
Would be two individual PRs.

buldo, 2024-10-07:
@EvilOlaf what about u-boot DTS?  
In my branch radxa u-boot used and patched

EvilOlaf, 2024-10-08:
I think here: https://github.com/armbian/build/tree/main/patch/u-boot/legacy

ceccocats, 2024-10-08:
Hi @buldo
I have a Core3566 (with wifi and emmc), i can help you testing.

buldo, 2024-10-08:
@ceccocats I made PR #7354
I don't know is armbian CI for PR provides images or not. I can share image for SD card that I built earlier( if it allowed here to post links on mega)