armbian/build #7304: [Bug]: The CCACHE_TEMPDIR variable is not working correctly!

The-going, 2024-09-27:
### What happened?

By default, CCACHE_TEMPDIR=$CCACHE_DIR/tmp and there is a lot of traffic in this folder.
During compilation, temporary files are created for each compilation target and the file is deleted at the end of the process. Thus, the size of this folder is not large and can be safely placed in tmpfs (RAM).
To increase the speed of compilation, this CCACHE_TEMPDIR folder must be placed in RAM.

What do I see today?
This variable is hard-coded and cannot be overridden:
```
lib/functions/main/start-end.sh:56:     declare -g -x CCACHE_TEMPDIR="${WORKDIR}/ccache_tmp" # Export CCACHE_TEMPDIR, under Workdir, which is hopefully under tmpfs. Thanks @the-Going for this.
```
- First of all, my nickname is @The-going
- Secondly, the code that I added allowed to redefine the location of the temporary compiler cache folder at the user's request, since the settings for mounting tmpfs in the user's OS are not known in advance.

### How to reproduce?

```bash
./compile.sh CCACHE_TEMPDIR="/tmp/ccachetmp" kernel
```
And see the log or try to find the folder in /tmp

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2024-09-28:
Hmm. It looks to me it works as intended (causes ccache to write temporary files to a certain place), but the way you want to use it (_change_ the place from the command-line) is not supported.

If you build with `SHOW_CCACHE=yes` more logs will be produced that may help with understanding.

> ```shell
> ./compile.sh CCACHE_TEMPDIR="/tmp/ccachetmp" kernel
> ```

I'm unsure this will do anything...

`CCACHE_TEMPDIR` is always overwritten (in `prepare_host_init()` as you pointed out) and setting it as a command line param probably does nothing. I'm unsure which runs first; building with `DEBUG=yes SHOW_CCACHE=yes` might reveal more.

Furthermore: since this was implemented, we've now a generic `TEMPDIR` variable that _should_ be picked up by ccache (and many others) if `CCACHE_TEMPDIR` wasn't set. Maybe we could simply drop all this handling.

> * First of all, my nickname is @The-going

I'm terribly sorry, no offense was intended.

The-going, 2024-09-29:
Good health, Ricardo.
Thanks for the comment, but I've already figured it out.

> I'm terribly sorry, no offense was intended.

There is no resentment.
I'm sorry that I expressed myself incorrectly.
The lack of documentation or a short explanation of the algorithm in the code is confusing.

There are two approaches to how to use tmpfs.
I use mounting the /tmp folder as tmpfs (/etc/fstab OS) and create temporary folders in it for current tasks.
You're taking full control of the situation. That is, you create temporary folders and mount them as tmpfs.
I like this approach better.

> Furthermore: since this was implemented, we've now a generic `TEMPDIR` variable that _should_ be picked up by ccache (and many others) if `CCACHE_TEMPDIR` wasn't set. Maybe we could simply drop all this handling.

It's definitely not worth giving up on this!
But the question remains. How do I properly split the mount in tmpfs or in a simple temporary folder of the file system, depending on the size that the task requires?
That is, if the RAM is small.

The-going, 2024-09-30:
I think it can be closed.

closed 2024-09-30 by The-going