armbian/build #7363: [Feature Request]: Add BananaPi R4 | BPI-R4

HeyMeco, 2024-10-12:
### Which feature would you like to have?

This issue should be more seen as a placeholder and discussion thread for adding support for Filogic 880 / MT7988A which is being used on BPI-R4. I will try to add support to Armbian but am also open for any suggestions and help as this will be my first board project to add Armbian support to.

There's already mainline kernel (edge, haven't tried this) and u-boot support too.

Most of the needed work can be found on @frank-w profile like the uEnv config:
https://github.com/frank-w/u-boot/blob/2024-04-bpi/uEnv_r4.txt
and a working kernel to add initial support:
https://github.com/frank-w/BPI-Router-Linux/tree/6.11-main

Here are links for the BSP:
6.1: https://github.com/BPI-SINOVOIP/BPI-R4-bsp-6.1
5.4: https://github.com/BPI-SINOVOIP/BPI-R4-bsp-5.4

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

sputnik2019, 2024-10-14:
@HeyMeco Thank you for your contribution，Here's some information about Armbian support for R3/R4：
https://github.com/BPI-SINOVOIP/armbian-build/commit/344a1aaf3871b1626c0cb9f544e591a549d1393e
But there are still some big bugs in handling uboot and boot file system partitions.

Thank you again for your selfless contribution！！

HeyMeco, 2024-10-14:
@sputnik2019 thank you for linking me to the commit. This will make it a lot easier for me to adapt it.

leggewie, 2025-08-21:
This has been completed, right?

Time to close?