armbian/build #7374: xiaomi-elish ubuntu oracular build failed

DOUBLEU9264, 2024-10-14:
### What happened?

[image]

### How to reproduce?

`./compile.sh build BOARD=xiaomi-elish BRANCH=current BUILD_DESKTOP=yes BUILD_MINIMAL=no DESKTOP_APPGROUPS_SELECTED='multimedia office remote_desktop' DESKTOP_ENVIRONMENT=gnome DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base KERNEL_CONFIGURE=no RELEASE=oracular`

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/jicequlege

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-10-14:
```
[🐳|🌱] Upgrading base packages [ Armbian ]
[🐳|🔨]   (Reading database ... 6772 files and directories currently installed.)
[🐳|🔨]   Preparing to unpack .../base-files_24.5.1-13.1ubuntu1-oracular_arm64.deb ...
[🐳|🔨]   dpkg-divert: error: 'diversion of /lib32 to /lib32.usr-is-merged by base-files' clashes with 'diversion of /lib32 to /.lib32.usr-is-merged by base-files'
[🐳|🔨]   dpkg: error processing archive /var/cache/apt/archives/base-files_24.5.1-13.1ubuntu1-oracular_arm64.deb (--unpack):
[🐳|🔨]    new base-files package pre-installation script subprocess returned error exit status 2
[🐳|🔨]   Errors were encountered while processing:
[🐳|🔨]    /var/cache/apt/archives/base-files_24.5.1-13.1ubuntu1-oracular_arm64.deb
[🐳|🔨]   E: Sub-process /usr/bin/dpkg returned an error code (1)
[🐳|🚸] Command failed, retrying in 5s [ chroot_sdcard_apt_get upgrade ]
```
This seems to be upstream bug related to Oracular. Can you use any other userspace?

igorpecovnik, 2024-10-14:
Yes, this Oracular problem was introduced few weeks ago ...

DOUBLEU9264, 2024-10-14:
it could build oracular successfully with empty config property `DESKTOP_APPGROUPS_SELECTED`, system works fine for now.

rpardini, 2024-10-15:
Two parts to this.

----

One, oracular is shipping base-files `13.3ubuntu6` right now. 
See OP's logs at line 131.
During "Upgrading base packages" (logs line 1135), _Armbian repos are enabled_, and contain a junk/old version from 24.5's release. "Upgrading" to it fails, of course, due to /usr merge (see below), but could be anything.

We've already tried ([once](https://github.com/armbian/build/pull/7123), [twice](https://github.com/armbian/build/pull/7313)) to disable Armbian's repos during rootfs build to avoid this chicken-egg problem. It should be disabled for rootfs cli & rootfs artifact, but now this needs handling when rootfs is built "inline" with image, as is the case for most users building on command line. Except here we've to handle the transition from building-rootfs to building-image. I sincerely think we should remove all repo-adding code from rootfs so SKIP_ARMBIAN_REPO is always equivalent to "yes" there no matter what.

Alternatively: remove the junk from the repo, or update it.

Long-term, we should split the repos (dists) so that the armbian/build produced packages don't ever get mixed with other stuff, and can be enabled selectively (eg NOT during rootfs build).

@DOUBLEU9264: you can build with `SKIP_ARMBIAN_REPO=yes` in the meantime to workaround this.

----

Two, I couldn't find much on the Ubuntu side, but https://wiki.debian.org/UsrMerge explains a bit. Recently the move moratorium has been [lifted](https://lists.debian.org/debian-devel-announce/2023/10/msg00003.html). 

Either way this (very complex) move is pivoting around `base-files` -- we should re-think we if really need to ship our own `base-files`, which unless I'm very forgetful is only for cosmetic purposes.

igorpecovnik, 2025-08-26:
> this fails in a different way now

Yeah, but original issue is not relevant anymore. This one is for @amazingfate when finds time.