armbian/build #7381: [Bug]:  kernel suffix mismatch in packaging scripts when used with PREEMPT_RT

tw-ilson, 2024-10-15:
### What happened?

When compiling the RK3588 Collabora kernel with PREEMPT_RT, the packaging scripts fail to recognize the changed kernel name:

```
### kernel_package.log 
--> (2830) INFO: Packaging kernel [ rockchip-rk3588 linux-rockchip-rk3588-collabora ]
--> (2830) INFO: Kernel .deb package version [ 6.9.0-S23bb-D558b-P47f0-C1eeeH5c21-HK01ba-Vc222-Ba3b7-R448a ]
--> (2830) COMMAND: rm -rf /home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot/*.old
--> (2830) INFO: Packaging linux-image [ rockchip-rk3588 linux-rockchip-rk3588-collabora ]
--> (2830) COMMAND: tree -C --du -h /home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot
   [ 39M]  /home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot
   ├── [219K]  config-6.9.0-rt5-collabora-rockchip-rk3588
   ├── [6.3M]  System.map-6.9.0-rt5-collabora-rockchip-rk3588
   └── [ 33M]  vmlinuz-6.9.0-rt5-collabora-rockchip-rk3588
   
     39M used in 0 directories, 3 files
--> (2830) INFO: Kernel-built image filetype [ vmlinuz-6.9.0-collabora-rockchip-rk3588: cannot open `/home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot/vmlinuz-6.9.0-collabora-rockchip-rk3588' (No such file or directory) ]
--> (2830) COMMAND: ls -la /home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot /home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot/vmlinuz-6.9.0-collabora-rockchip-rk3588
   ls: cannot access '/home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot/vmlinuz-6.9.0-collabora-rockchip-rk3588': No such file or directory
   /home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot:
   total 40280
   drwxrwxr-x 2 root root      100 Oct 15 20:55 .
   drwxrwxr-x 3 root root       60 Oct 15 20:09 ..
   -rw-rw-r-- 1 root root   224121 Oct 15 20:55 config-6.9.0-rt5-collabora-rockchip-rk3588
   -rw-rw-r-- 1 root root  6596243 Oct 15 20:55 System.map-6.9.0-rt5-collabora-rockchip-rk3588
   -rw-rw-r-- 1 root root 34421248 Oct 15 20:55 vmlinuz-6.9.0-rt5-collabora-rockchip-rk3588
--> (2830) ERROR: Error 2 occurred in main shell [ at /home/tom/build/lib/functions/logging/runners.sh:211
       run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
           run_host_command_logged() --> lib/functions/logging/runners.sh:193
      kernel_package_callback_linux_image() --> lib/functions/compilation/kernel-debs.sh:221
                 create_kernel_deb() --> lib/functions/compilation/kernel-debs.sh:131
      prepare_kernel_packaging_debs() --> lib/functions/compilation/kernel-debs.sh:72
                    kernel_package() --> lib/functions/compilation/kernel.sh:248
                     do_with_hooks() --> lib/functions/general/extensions.sh:567
                   do_with_logging() --> lib/functions/logging/section-logging.sh:81
      kernel_prepare_build_and_package() --> lib/functions/compilation/kernel.sh:171
                    compile_kernel() --> lib/functions/compilation/kernel.sh:100
      artifact_kernel_build_from_sources() --> lib/functions/artifacts/artifact-kernel.sh:240
       artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
          obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:280
             do_with_default_build() --> lib/functions/main/default-build.sh:42
                  cli_artifact_run() --> lib/functions/cli/cli-artifact.sh:72
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                              main() --> compile.sh:50
    ]
   -->--> command failed with error code 2 after 0 seconds
--> (2830) WARNING: stacktrace for failed command [ exit code 2:/usr/bin/env bash -e -o pipefail -c ls -la /home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot /home/tom/build/.tmp/work-27fcaccd-d9e9-4ae5-aee2-59728cf75839/kernel_dest_install_dir-GUlIt/image/boot/vmlinuz-6.9.0-collabora-rockchip-rk3588
           run_host_command_logged() --> lib/functions/logging/runners.sh:193
      kernel_package_callback_linux_image() --> lib/functions/compilation/kernel-debs.sh:221
                 create_kernel_deb() --> lib/functions/compilation/kernel-debs.sh:131
      prepare_kernel_packaging_debs() --> lib/functions/compilation/kernel-debs.sh:72
                    kernel_package() --> lib/functions/compilation/kernel.sh:248
                     do_with_hooks() --> lib/functions/general/extensions.sh:567
                   do_with_logging() --> lib/functions/logging/section-logging.sh:81
      kernel_prepare_build_and_package() --> lib/functions/compilation/kernel.sh:171
                    compile_kernel() --> lib/functions/compilation/kernel.sh:100
      artifact_kernel_build_from_sources() --> lib/functions/artifacts/artifact-kernel.sh:240
       artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
          obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:280
             do_with_default_build() --> lib/functions/main/default-build.sh:42
                  cli_artifact_run() --> lib/functions/cli/cli-artifact.sh:72
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                              main() --> compile.sh:50 ]
```

### How to reproduce?

```
cd /userpatches/kernel/rockchip-rk3588-collabora/
curl -LO https://cdn.kernel.org/pub/linux/kernel/projects/rt/6.9/patch-6.9-rt5.patch.gz
gunzip patch-6.9-rt5.patch.gz
cd ~/build
./compile.sh kernel BOARD=orangepi5 BRANCH=collabora 
```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/coluleleka

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

royka1, 2024-12-01:
Remove the part from the patch where it adds the localversion file or use 6.12 which doesn't need a patch.

leggewie, 2025-08-21:
> collabora kernel branch could be removed alltogether. It is abandoned.

are you suggesting to do this only for this board / family or in general?

It appears to me that the following files might be affected:
config/boards/radxa-nio-12l.conf
config/sources/families/genio.conf
config/sources/families/rockchip-rk3588.conf

The radxa-nio-12l in particular currently only lists the collabora kernel for KERNEL_TARGET.

leggewie, 2025-08-22:
OK, let's limit it to config/sources/families/rockchip-rk3588.conf, then.