armbian/build #7399: Failed to mount binfmt_misc

clarkmcc, 2024-10-21:
### What happened?

When building, I'm getting an error that we can't mount binfmt_misc.

I'm running off my own branch based on [bca58a204ec6046ba36d8517c56585513dc75880](https://github.com/armbian/build/commit/bca58a204ec6046ba36d8517c56585513dc75880) that contains only userpatches.

I ran this and didn't get any output so I'm assuming it worked.
```
root@buildbox:/home/owner/actions-runner-1# modprobe binfmt_misc
```

What am I missing, and what can I check next?

### How to reproduce?

See the attached log file for the exact command I'm running

### Branch

other (only relevant in corner cases)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/iwosaniceg

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

clarkmcc, 2024-10-21:
```
root@buildbox:/home/owner/actions-runner-1# cat /boot/config-$(uname -r) | grep CONFIG_BINFMT_MISC
CONFIG_BINFMT_MISC=m
```

clarkmcc, 2024-10-21:
I rebased on top of [549f321ebf61afac6bdee5b8966fa358c7136267](https://github.com/armbian/build/commit/549f321ebf61afac6bdee5b8966fa358c7136267) (latest commit to v24.08) and am still getting the same issue.

leggewie, 2025-08-19:
Thank you for reporting back.  Apologies for picking this up so late.  I will close the ticket now.  Please feel free to report any issue you have in the future.