armbian/build #7418: [Bug]: [🔨]   E: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed

doofin, 2024-10-29:
### What happened?

Hi, I got this err but verified that I have installed gpgv
[image]
[image]

### How to reproduce?

[✨] Repeat Build Options (early) [ ./compile.sh build BOARD=orangepi5 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm ]

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-10-30:
full logs missing

closed 2024-11-06 by EvilOlaf

melodie11, 2025-03-10:
Hello,
I happen to get this message in the console below the Ubiquity installer, while installing a newly built Ubuntu Remix Openbox. (x86_64 arch - see linuxvillage.org/en, if curious)
I ensured to have gpgv and gpgv1 installed before rebuilding the ISO (using Cubic), but I still get the message when doing a test install in Virtualbox (the message occurs in the terminal which comes with the graphical Ubiquity installer).

However, it is not blocking.