armbian/build #7451: [Bug]: Armbian 24.8.4 bookworm: Network forwarding doesn't work

jeje232, 2024-11-07:
### What happened?

lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Armbian 24.8.4 bookworm
Release:	12
Codename:	bookworm

uname -a
Linux xxxxxx-pre 6.6.44-current-sunxi64 #1 SMP Sat Aug  3 06:54:42 UTC 2024 aarch64 GNU/Linux

OrangePI Zero v2

Network forwarding doesn't work

### How to reproduce?

ip -4 addr show scope global
2: end0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    inet 192.168.1.135/24 metric 100 brd 192.168.1.255 scope global dynamic end0
       valid_lft 587718sec preferred_lft 587718sec

iptables -A PREROUTING -t nat -i lo -p udp --dport 2222 -j DNAT --to 127.0.0.1:22
iptables -A PREROUTING -t nat -i lo -p tcp --dport 2222 -j DNAT --to 127.0.0.1:22
iptables -A FORWARD -p tcp -d 127.0.0.1 --dport 2222 -m state --state NEW,ESTABLISHED,RELATED -j ACCEPT
iptables -A FORWARD -m limit --limit 3/minute --limit-burst 3 -j LOG --log-prefix "IPT FORWARD packet"
iptables -A POSTROUTING -t nat -j MASQUERADE -o lo
iptables -A INPUT -m state --state RELATED,ESTABLISHED -j ACCEPT
iptables -A INPUT -j ACCEPT
iptables -A OUTPUT -j ACCEPT
iptables -A FORWARD -j ACCEPT

ssh -l root -p 22 localhost => OK
ssh -l root -p 2222 localhost => KO

iptables -A INPUT -j ACCEPT
iptables -A OUTPUT -j ACCEPT
iptables -A FORWARD -j ACCEPT
iptables -A PREROUTING -t nat -i end0 -p udp --dport 2222 -j DNAT --to 192.168.1.135:22
iptables -A PREROUTING -t nat -i end0 -p tcp --dport 2222 -j DNAT --to 192.168.1.135:22
iptables -A FORWARD -p tcp -d 192.168.1.135 --dport 2222 -m state --state NEW,ESTABLISHED,RELATED -j ACCEPT
iptables -A FORWARD -m limit --limit 3/minute --limit-burst 3 -j LOG --log-prefix "IPT FORWARD packet"
iptables -A POSTROUTING -t nat -j MASQUERADE -o end0
iptables -A INPUT -m state --state RELATED,ESTABLISHED -j ACCEPT

ssh -l root -p 22 192.168.1.135 => OK
ssh -l root -p 2222 192.168.1.135 => KO

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jeje232, 2024-11-07:
Everything com from the trying to create a docker compose project:
version: '3.6'
services:
  db:
    image: mariadb:10.3
    environment:
      MYSQL_ROOT_PASSWORD: secret
  phpmyadmin:
    image: phpmyadmin
    restart: always
    ports:
      - 8080:80
    environment:
      - PMA_HOSTS=db

# FROM test-db-1 => test-db-1 3306
docker exec -ti test-db-1 bash -c 'timeout 0.5 echo -n 2>/dev/null < /dev/tcp/test-db-1/3306 && echo "open" || echo "closed"'
-----------
open
-----------

# FROM test-phpmyadmin-1 => test-db-1 3306
docker exec -ti test-phpmyadmin-1 bash -c 'timeout 0.5 echo -n 2>/dev/null < /dev/tcp/test-db-1/3306 && echo "open" || echo "closed"'
-----------
closed
-----------

closed 2024-11-07 by EvilOlaf

igorpecovnik, 2024-11-07:
... and reproduce this on mainstream x86 hardware with mainline kernel.