armbian/build #7482: [Bug]: rockchip-rk3588.conf overrides OVERLAY_PREFIX

bmx666, 2024-11-15:
### What happened?

In file `config/sources/families/rockchip-rk3588.conf`

```
OVERLAY_PREFIX='rockchip-rk3588'
```

It must use default definition only, without overriding
```
OVERLAY_PREFIX="${OVERLAY_PREFIX:-"rockchip-rk3588"}" # default to 'rockchip-rk3588' if not set by board
```

### How to reproduce?

define custom `OVERLAY_PREFIX` in config/boards that depends on `rockchip-rk3588`.

For example, `config/boards/rock-5b.conf`

```
BOARDFAMILY="rockchip-rk3588"
OVERLAY_PREFIX="rock-5b"
```

In generated image, in file `/boot/armbianEnv.txt`

```
overlay_prefix=rockchip-rk3588
```

But it must be
```
overlay_prefix=rock-5b
```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leggewie, 2025-08-19:
I don't understand overlays enough.  @igorpecovnik Can you confirm that this is indeed an issue?

bmx666, 2025-08-19:
@leggewie 

It's the same patch that was made a long time ago here at `config/sources/families/include/rockchip64_common.inc` in commit -> https://github.com/armbian/build/commit/eefbe3fa6c865617660b20b263767a56cb7f59a7

https://github.com/armbian/build/blob/4b50772747367d817b9e866015e57f9284aebd69/config/sources/families/include/rockchip64_common.inc#L16

leggewie, 2025-08-25:
Thank you for that explanation.  That makes perfect sense indeed.

leggewie, 2025-08-25:
The correct solution is actually to remove the line from config/sources/families/rockchip-rk3588.conf.  That way, the value is either set in the board config file or as a fallback in the included common.inc file.

leggewie, 2025-08-26:
> I disagree that this is the correct solution. In Armbian the standard is that all overlays are by cpu family. There shouldn't be cases where there are board specific overlay_prefixes.

Thank you for sharing your perspective, @SteeManMI. Much appreciated.

@bmx666 What is your use-case for doing what you are doing?

I was looking at things purely from the perspective of logic, as laid out by @bmx666 which I found compelling in that if you override a variable in your board config, you'd expect it to stick and it currently doesn't.  He certainly has a very valid point in that Armbian code currently is not consistent in that `config/sources/families/include/rockchip64_common.inc` very specifically allows for board-level overrides to the OVERLAY_PREFIX variable.  @bmx666 quoted the comment documenting this.  That goes directly against your statement and ought to be fixed if that is the consensus.

bmx666, 2025-08-26:
> What is your use-case for doing what you are doing?

I made overlay fix for customers. Because they could have the same overlay suffix like

* boardname-r1-acc-sensor
* boardname-r1-light-sensor
* boardname-r1-serial-gps
* boardname-r2-acc-sensor
* boardname-r2-serial-gps
* ...

For example, boards should have the same overlay suffixes, but PCB can be different for every revisions, like HW could be for on SPI or I2C, or on different buses on rev1 some sensors on i2c3/4, but on rev2 on i2c5/i2c7. Or use different GPIOs, but for similar HW, like enable/disable power.

To abstract business logic of HW from PCB, I made the suffixes the same to simplify and avoid confusion.

But in the case of hardcoded "rockchip-rk3588" it doesn't make sense, as it was done for other processors. It is not clear why someone made such a decision and separated RK3588 from other Rockchip.

leggewie, 2025-08-26:
> But in the case of hardcoded "rockchip-rk3588" [...]. It is not clear why someone made such a decision and separated RK3588 from other Rockchip.

Let's find out more.

@SteeManMI I tried to find those messages in the forum you mentioned and did not manage to do so, so far.  Am I correct to think you are referring to thread 29202?  Around what page or time?

bmx666, 2025-08-26:
I briefly read post, but I didn't notice to issue `OVERLAY_PREFIX` for rk3588.

I already saw the same problem more than 10 years ago.

1. (COMMON) Some people want to minimize the number of DTBO for beginners, for example by enabling i2c8, i2c3, spi2, and so on. Go to Aliexpress and buy 1000+ i2c/spi sensors, add overlays, and everything works right away. It's a RasperryPi-like approach and similar clones with PINOUT blocks.

2. (BOARD SPECIFIC) But there is a big problem – boards in Armbian based on Rockchip have only 10-20% in common, PINMUX is different, not all pins are out, have PCIe or etc. And I'm not surprised that people on the forum just grab the DTBO from one board and try to use it for another without even knowing that the board doesn't have i2c8, or that the sensors are connected to spi1 and not spi0. And yes, this approach requires basic knowledge of DTBO, but the problems come from the Linux kernel itself because of the high knowledge barrier you have to overcome before you can pass the wall.

Both sides have pros and cons. But at the end you ALWAYS get exceptions like now you must use `i2c-light-sensor-V2.dtbo`, then it will be splitted into `i2c-light-sensor-rpi2.dtbo` and `i2c-light-sensor-rpi3.dtbo`.
And no one stop you from using symbolic links:
* `boardname-r1-enable-i2c2.dtbo` -> `common-CPU-enable-i2c2.dtbo`
* `boardname-r2-enable-i2c2.dtbo` -> `common-CPU-enable-i2c2.dtbo`

Anyway, as a professional developer, I don't like the RasperryPi-like approach at all because it leads to more complex and confusing solutions where people don't even try to understand “WHAT THE HELL IS GOING ON HERE?” and use their brain cells. And documentation about DTS/DTBO should be spread around the world by explaining how to read, build, and use it. Instead of taking a random DTBO and hoping it works.

bmx666, 2025-08-28:
> we have your PR sitting open forever

@leggewie I have already proposed a solution without breaking anything - https://github.com/armbian/build/issues/7482#issuecomment-3200395954. If someone decides to add a custom board based on RK3588 and wants custom overlays instead of the hardcoded “rockchip-rk3588-XXX.dtbo,” they will encounter the same problem I did.
I spent several hours trying to solve this problem, so I don't think it will hurt anyone, but only save time and avoid headaches.

SteeManMI, 2025-08-28:
@bmx666 Wading more into this conversation.  I've reread this whole PR and what is missing is an explaination on why you want the overlay prefix to be "rockpi-5b".  After all this is just a naming convention and any dtbo's could be named according to any prefix.  Why are you insisting that you need to override the default value?  I'm sure you have a reason, but I don't see it explained here.

SteeManMI, 2025-08-28:
But as was stated earlier, the standard Armbian functionality is that overlay_prefix is by family.  So you are just pointing out other boards that are attempting to do things wrong.  That doesn't explain what functionality you are trying to gain/fix with a changed overlay_prefix.  So please explain what underlying issue you hope to address with using a different overlay_prefix than is currently in place.  You aren't explaining the end functionality that you are trying to get to, but instead only proposing a specific implementation that you think is correct.  But there may be multiple ways to accomplish your end goal if that goal is explained.

bmx666, 2025-08-28:
Just a final note — I don’t want to go too deep into details, but I’ve seen several companies ship their own custom vendor kernels and U-Boot with hardcoded DTS prefixes, for example `firefly-..., insoma-..., cobex-z1-..., cobex-z2-..., etc.` Some Chinese suppliers also provide kernel DTS files like r`k3588-myboard` or `myboard-rk3588`. It quickly becomes messy to fix, move, and align everything.

I don’t have the time to spend 100+ hours figuring out all the variations being discussed here, and I haven’t seen a clear solution. From my perspective, if someone simply wants to set `OVERLAY_PREFIX=cobex-z2` for any board, then that should just work — and it saves everyone a lot of time.

closed 2025-08-28 by bmx666

leggewie, 2025-08-29:
> How do you then explain the commit above that [@bmx666](https://github.com/bmx666) kindly referenced that is very, very explicit in going the exact opposite direction?

Answering myself here after a little bit of code inspection and still lacking a 100% clear picture:  The difference between config/sources/families/include/rockchip64_common.inc and config/sources/families/rockchip-rk3588.conf appears to be that the former obviously being an include file, used by other family definitions but not by a single board directly.  So, the chain is then for example config/boards/rock-5b.conf -> config/sources/families/rockchip-rk3588.conf -> config/sources/families/include/rockchip64_common.inc

In that case, I would concede that the wording in config/sources/families/include/rockchip64_common.inc is at least misleading and should probably be changed to "# default to 'rockchip' if not set by board _family_".

@SteeManMI Did I get this right?

amazingfate, 2025-08-29:
The doc says we support board level overlay_prefix: https://github.com/armbian/build/blob/main/config/boards/README.md?plain=1#L103
And there is one family supporting it: https://github.com/armbian/build/blob/main/config/sources/families/sun8i.conf#L12-L14
While most families don't support this feafure.